iNavFlight/inav #9041: [6.1-RC1] (Auto)-throttle icon analog OSD

0crap, 2023-05-07:
## Current Behavior
(Auto)-throttle icon on OSD shows weird "I" sign in front of it.
Using ANALOG OSD.
(And yes, I updated my fonts on v6.1 RC1 with lipo connected.)

## Steps to Reproduce
1. Add (Auto) throttle icon to analog OSD

## Expected behavior
Just show the S icon as it was before.
The weird I sign is just unwanted clutter.

## Suggested solution(s)
Remove large I sign from the auto-throttle OSD element.

## Additional context
Picture:
[image]

---

# version
# INAV/MATEKF405 6.1.0 Apr 21 2023 / 17:41:41 (2d7a057d)
# GCC-10.2.1 20201103 (release)

- FC Board name and vendor: MATEKF405
- INAV version string: 6.1 RC1

MrD-RC, 2023-05-07:
That is showing that is is the scaled throttle.

closed 2023-05-07 by MrD-RC

0crap, 2023-05-07:
OK, thx.
Still unwanted but OK, it is what it is.