iNavFlight/inav #9042: Mate Airspeed MS4525 no readings

przemekgalazka, 2023-05-07:
on Inav 4.1 all good so hardware is OK.

ON INAV 6.1 I can see that airsensor is found but readings on sensors tabs are not changing at all (IAS)
Sensor is connected via I2C number two (with stock matek cable)

Tested on Matek F765-WSE

mmosca, 2023-05-09:
Probably fixed by #9019

mmosca, 2023-05-09:
@przemekgalazka Can you verify that this hex fixes your issue?

[inav_6.1.0_MATEKF765SE.zip](https://github.com/iNavFlight/inav/files/11430277/inav_6.1.0_MATEKF765SE.zip)

M0j0Risin, 2023-05-09:
> @przemekgalazka Can you verify that this hex fixes your issue?
> 
> [inav_6.1.0_MATEKF765SE.zip](https://github.com/iNavFlight/inav/files/11430277/inav_6.1.0_MATEKF765SE.zip)

I didn't try your zip, but I can confirm that building off the current "release_6.1.0" branch seems to resolve the issue for me. That is, the airspeed value was always 0 on 6.1 RC1 and now displays non-zero values when I blow on the sensor on the bench after flashing with a locally compiled release_6.1.0 binary. So I think the changes Roman made in #9019 did indeed resolve the issue.  Mine is a H743-WLITE, but I doubt that makes a difference.

mmosca, 2023-05-10:
> I didn't try your zip, but I can confirm that building off the current "release_6.1.0" branch seems to resolve the issue for me. That is, the airspeed value was always 0 on 6.1 RC1 and now displays [...]

Thank you for testing.

The hex I posted was also compiled from the latest release_6.1.0.

@przemekgalazka It should be fixed for you too. 6.1 will be released soon, meanwhile you can use the hex I posted.

closed 2023-05-10 by mmosca

przemekgalazka, 2023-05-10:
thank you for a prompt solution :)

I will check it out but I do not want flash my F765 yet as it is now in my jet :)
I can test it on H743
Could you give me a hex for matek H743? Compiling this out of source code is not the case for me yet :)

przemekgalazka, 2023-05-11:
I built hex  for H743 from master and it seems that it works (v7.0.0-snapshot)
:)
well done !

ps. How do you build 6.1-RC1 with such bug fixes ? is there a branch for 6.1-RC2 or for now it is all merged only to master? @DzikuVx ?

mmosca, 2023-05-11:
I did not fix the issue. :)

Anyway, you can use release_6.1.0 branch, but the release will be coming out very soon.