iNavFlight/inav #9052: Multirotor waypoint speed setting issue

Jetrell, 2023-05-16:
When a velocity is set for each waypoint. The speed value (700cm/s in this case) was converted to a hold time for that waypoint.
The mission file checks out fine. And will run if `nav_auto_speed` is used in place of the individual speed set points,

[image]

[image]

- FC Board name and vendor: Mamba F405
- INAV version string: 6.0.0

Jetrell, 2023-05-18:
I did have `Enforce altitude at waypoint` enabled. Not sure why. I never use this feature.
Looking at the DVR. It was within 1m of the target altitude. And never tried to get any closer, in the time I let it hold the WP.

These are the related WP settings. 
```
nav_wp_load_on_boot = ON
nav_wp_radius = 300
nav_wp_enforce_altitude = 200
nav_wp_max_safe_distance = 600
nav_fw_wp_tracking_accuracy = 5
nav_fw_wp_turn_smoothing = OFF
nav_auto_speed = 1806
nav_max_auto_speed = 2000
```
Can you contact me on Discord and I'll send you the mission file.

breadoven, 2023-05-19:
> I did have `Enforce altitude at waypoint` enabled. Not sure why. I never use this feature. Looking at the DVR. It was within 1m of the target altitude. And never tried to get any closer, in the time [...]

I was just wondering what waypoint actions are set and the corresponding parameter1 ? Any POSHOLD_TIME waypoints set or are they all standard WAYPOINT types ?

Jetrell, 2023-05-19:
It has 13 marker... 12 Waypoints and 1 Landing at the end.
Each marker has a slightly different Altitude and Speed set points... `PH_time` was not used at all in this mission.

[image]

stronnag, 2023-05-19:
This, suitably relocated to protect the innocent.
```
<mission>
  <version value="2.3-pre8" />
  <missionitem no="1" action="WAYPOINT" lat="35.761" lon="140.378945" alt="10" parameter1="700" parameter2="0" parameter3="0" flag="0" />
  <missionitem no="2" action="WAYPOINT" lat="35.7606358" lon="140.3788572" alt="12" parameter1="900" parameter2="0" parameter3="0" flag="0" />
  <missionitem no="3" action="WAYPOINT" lat="35.760294" lon="140.3788317" alt="12" parameter1="700" parameter2="0" parameter3="0" flag="0" />
  <missionitem no="4" action="WAYPOINT" lat="35.760103400000006" lon="140.3791723" alt="12" parameter1="700" parameter2="0" parameter3="0" flag="0" />
  <missionitem no="5" action="WAYPOINT" lat="35.7600574" lon="140.3794352" alt="12" parameter1="700" parameter2="0" parameter3="0" flag="0" />
  <missionitem no="6" action="WAYPOINT" lat="35.759540900000005" lon="140.3801547" alt="18" parameter1="700" parameter2="0" parameter3="0" flag="0" />
  <missionitem no="7" action="WAYPOINT" lat="35.75868800000001" lon="140.3801083" alt="15" parameter1="700" parameter2="0" parameter3="0" flag="0" />
  <missionitem no="8" action="WAYPOINT" lat="35.758431300000005" lon="140.3798697" alt="15" parameter1="2000" parameter2="0" parameter3="0" flag="0" />
  <missionitem no="9" action="WAYPOINT" lat="35.7592787" lon="140.3785313" alt="17" parameter1="700" parameter2="0" parameter3="0" flag="0" />
  <missionitem no="10" action="WAYPOINT" lat="35.7595095" lon="140.37821549999998" alt="34" parameter1="700" parameter2="0" parameter3="0" flag="0" />
  <missionitem no="11" action="WAYPOINT" lat="35.7602625" lon="140.378322" alt="32" parameter1="1500" parameter2="0" parameter3="0" flag="0" />
  <missionitem no="12" action="WAYPOINT" lat="35.760497900000004" lon="140.3787565" alt="10" parameter1="1000" parameter2="0" parameter3="0" flag="0" />
  <missionitem no="13" action="LAND" lat="35.7611568" lon="140.37892829999998" alt="10" parameter1="0" parameter2="0" parameter3="0" flag="165" />
</mission>

```

breadoven, 2023-05-19:
OK I can what the problem is now, it is a bug. Fix to come.

closed 2023-05-23 by breadoven