iNavFlight/inav #9057: Battery continuation / Persist battery mAh across sessions

iceboundflame, 2023-05-20:
betaflight recently introduced a "battery continuation" feature to remember the mAh consumed from the previous session, if booted with a partially discharged battery. Does inav support something similar?

Knowing mAh consumed/remaining is important for long range flight. When flying with larger batteries like Li-Ions, you may not finish a whole pack in one flight session, so manual bookkeeping would be required in order to track total mAh consumed on the battery.

MrD-RC, 2023-05-20:
I can see that this can be useful. But, it is better to use cell voltage for remaining flight time. Battery capacity can change with time.

iceboundflame, 2023-05-20:
Thanks - it's true that capacity changes over time, but  it does so slowly over many cycles, and the amount of mAh put in is reported by the charger each time you charge the battery. The remaining flight time is much harder to predict from cell voltage, since the voltage does not drop linearly and follows a discharge curve. The voltage reading in flight is also heavily affected by voltage sag (though sag compensation can mitigate this last part).

In practice I like to use a combination of these factors to monitor my battery, knowing the critical voltage to land when fully depleted, but also using mAh capacity for longer range estimates.