iNavFlight/inav #9086: SITL MacOS reports invalid settings

mmosca, 2023-05-29:
Adding this here so we can keep track of it.

## Current Behavior
SITL build struggles with settings in macosx.

If I set nav_fw_pos_hdg_pidsum_limit to 350, fw_p_pitch also gets set to 350.

If I set fw_p_pitch to 5, nav_fw_pos_hdg_pidsum_limit gets set to 5.

## Steps to Reproduce

1. run INAV's SITL target on OSX.
2. enter cli
3. type status
4. get nav_fw_pos_hdg_pidsum_limit
5. set  nav_fw_pos_hdg_pidsum_limit=350
6. status 
7. get fw_p_pitch

## Expected behavior
Settings are not mixed up and saved propperly.

## Suggested solution(s)

## Additional context
List of value offsets in OSX

```
Value offset looptime: 0
Value offset gyro_hardware_lpf: 2
Value offset gyro_anti_aliasing_lpf_hz: 4
Value offset gyro_anti_aliasing_lpf_type: 6
Value offset gyro_main_lpf_hz: 8
Value offset gyro_main_lpf_type: 10
Value offset gyro_use_dyn_lpf: 11
Value offset gyro_dyn_lpf_min_hz: 12
Value offset gyro_dyn_lpf_max_hz: 14
Value offset gyro_dyn_lpf_curve_expo: 16
Value offset init_gyro_cal: 17
Value offset gyro_zero_x: 18
Value offset gyro_zero_y: 20
Value offset gyro_zero_z: 22
Value offset ins_gravity_cmss: 24
Value offset vbat_adc_channel: 0
Value offset rssi_adc_channel: 1
Value offset current_adc_channel: 2
Value offset airspeed_adc_channel: 3
Value offset acc_notch_hz: 16
Value offset acc_notch_cutoff: 17
Value offset acc_hardware: 0
Value offset acc_lpf_hz: 2
Value offset acc_lpf_type: 18
Value offset acczero_x: 4
Value offset acczero_y: 6
Value offset acczero_z: 8
Value offset accgain_x: 10
Value offset accgain_y: 12
Value offset accgain_z: 14
Value offset rangefinder_hardware: 0
Value offset rangefinder_median_filter: 1
Value offset opflow_hardware: 0
Value offset opflow_scale: 4
Value offset align_opflow: 1
Value offset align_mag: 4
Value offset mag_hardware: 8
Value offset mag_declination: 0
Value offset magzero_x: 10
Value offset magzero_y: 12
Value offset magzero_z: 14
Value offset maggain_x: 16
[... 209 lines cut ...]
Value offset nav_mc_vel_xy_p: 130
Value offset nav_mc_vel_xy_i: 132
Value offset nav_mc_vel_xy_d: 134
Value offset nav_mc_vel_xy_ff: 136
Value offset nav_mc_heading_p: 154
Value offset nav_mc_vel_xy_dterm_lpf_hz: 228
Value offset nav_mc_vel_xy_dterm_attenuation: 232
Value offset nav_mc_vel_xy_dterm_attenuation_start: 233
Value offset nav_mc_vel_xy_dterm_attenuation_end: 234
Value offset nav_fw_pos_z_p: 26
Value offset nav_fw_pos_z_i: 28
Value offset nav_fw_pos_z_d: 30
Value offset nav_fw_pos_xy_p: 34
Value offset nav_fw_pos_xy_i: 36
Value offset nav_fw_pos_xy_d: 38
Value offset nav_fw_heading_p: 66
Value offset nav_fw_pos_hdg_p: 82
Value offset nav_fw_pos_hdg_i: 84
Value offset nav_fw_pos_hdg_d: 86
Value offset nav_fw_pos_hdg_pidsum_limit: 10
```
---

- FC Board name and vendor: SITL
- INAV version string: 7.0.0

stronnag, 2023-05-29:
Is that all in your list, or did you just post up to `nav_fw_pos_hdg_pidsum_limit` ?

stronnag, 2023-05-29:
And FYI, Linux x86_64:

```
...
Value offset nav_fw_heading_p: 66
Value offset nav_fw_pos_hdg_p: 82
Value offset nav_fw_pos_hdg_i: 84
Value offset nav_fw_pos_hdg_d: 86
Value offset nav_fw_pos_hdg_pidsum_limit: 266
...
```

stronnag, 2023-05-29:
The Darwin example is 271 items. Linux & FreeBSD (GNU ld) generate 544 items (are are identical).

stronnag, 2023-05-29:
Consistent across architectures / OS as long as using GNU ld (with `sitl.ld`), Linux x86_64, ia32, riscsv64, freebsd x86_64, cygwin. The common factor is GNU ld and how it lays out the `pg*` sections iaw `sitl.ld`. 

How to beat the Darwin linker into ~submission~ doing the equivalent is key (imo at least).

stronnag, 2023-05-29:
> Is that all in your list, or did you just post up to `nav_fw_pos_hdg_pidsum_limit` ?

Gotya. It fails when it hits the first invalid item ....
```
Value offset nav_fw_heading_p: 66
Value offset nav_fw_pos_hdg_p: 82
Value offset nav_fw_pos_hdg_i: 84
Value offset nav_fw_pos_hdg_d: 86
Value offset nav_fw_pos_hdg_pidsum_limit: 10
Invalid settings for nav_fw_pos_hdg_pidsum_limit: min: 100 max: 1000
```

mmosca, 2023-05-29:
over here is seems like nav_fw_pos_hdg_pidsum_limit and fw_p_pitch are writting to the same place. I don't know if there are more issues. 

Slowly chipping away at debugging this, but still fighting to get lldb integraton in vs code to behave