iNavFlight/inav #9087: Increase navigation update rate for newer ublox GPS modules

mluessi, 2023-05-30:
## Current Behavior

When using a newer ublox GPS (e.g. M9, M10, F9P), the GPS update rate is configured to be only 5 Hz, while the hardware supports much higher rates. E.g. the F9P supports 18 Hz with 3 constellations, 25 Hz with two. Older ublox modules (M7, M8) use 10 Hz. 

## Steps to Reproduce
Connect ublox module, check rate 

## Expected behavior
The GPS navigation rate should be set higher if the hardware supports it.

## Suggested solution(s)
Set rate higher depending on ublox GPS being used

- FC Board name and vendor: NA
- INAV version string: NA

mmosca, 2023-06-06:
#9103 will set m9/m10 to 10Hz if configured to use UBLOX7.

I may add a way to configure the update rate for future proofing and supporting higher rates.

Increasing update rates may also affect the number of satellites a module can track which is the reason behind m9+ currently being set to 5Hz.

mmosca, 2023-06-09:
This will be available in 7.0

closed 2023-06-09 by mmosca

cmanley, 2023-10-30:
Is there a way to configure this in ublox u-center 2? I tried setting it to 10hz for my HGLRC M100 MINI for use in INAV 6.1 but it INAV still connects at 5hz. My guess is that this GNSS unit doesn't have flash memory.

mmosca, 2023-10-30:
You need inav 7, as inav 6 will auto configure it

cmanley, 2023-10-30:
Thank you, I understand now, and thanks for the MR too.