iNavFlight/inav #9091: I2C Errors: 5 on QMC5883L

D12A13, 2023-06-01:
## Current Behavior
First Thx for this great software and effort you are investing to make it avilable.
INAV show I2C Errors: 5 on SPEEDYBEE F7 V3 and a BLITZ M10 GPS with compass module QMC5883L.
The GPS Module and the mag compass are working fine, it just shows this error at the button line and at the status command.
I get sensor output on the sensor tab.
I tried I2C Speed 400 and 800 no change.
## Steps to Reproduce

1. Set Magnetometer to QMC5883 on SPEEDYBEE F7 V3 and an BLITZ M10 GPS

## Expected behavior

I2C Errors: 0

## Suggested solution(s)
On Betaflight I was able to fix this with set mag_i2c_device = 1, but this command is not avilable at INAV

## Additional context
status
INAV/SPEEDYBEEF7V3 6.1.0 May 11 2023 / 14:10:45 (d6ce6e70)
GCC-10.2.1 20201103 (release)
System Uptime: 262 seconds
Current Time: 2023-06-01T17:09:34.349+00:00
Voltage: 0.00V (1S battery - NOT PRESENT)
CPU Clock=216MHz, GYRO=BMI270, ACC=BMI270, BARO=BMP280, MAG=QMC5883
STM32 system clocks:
  SYSCLK = 216 MHz
  HCLK   = 216 MHz
  PCLK1  = 54 MHz
  PCLK2  = 108 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
SD card: Manufacturer 0xc8, 491520kB, 05/2165, v12.8, 'APPSD'
Filesystem: Ready
Stack size: 6144, Stack address: 0x20010000, Heap available: 1716
I2C Errors: 5, config size: 9031, max available config: 16384
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 2, used = none
   CURRENT : configured = ADC 3, used = ADC 3
  AIRSPEED : configured = none, used = none
System load: 19, cycle time: 515, PID rate: 1941, RX rate: 9, System rate: 9
Arming disabled flags: RX CLI
OSD: MSP DisplayPort: BetaFlight Compatability mode (HD) [53 x 20]
VTX: not detected

https://pastebin.com/9F4SnKnq
---

- FC Board name and vendor:
 SpeedyBee F7 V3 BL32 50A 30x30 Stack
- INAV version string:
INAV/SPEEDYBEEF7V3 6.1.0 May 11 2023 / 14:10:45 (d6ce6e70)
GCC-10.2.1 20201103 (release)

DzikuVx, 2023-06-02:
This is expected. Each check for I2C sensor that is not present is one I2C transaction that ends with an error.

closed 2023-06-02 by DzikuVx

D12A13, 2023-06-02:
> This is expected. Each check for I2C sensor that is not present is one I2C transaction that ends with an error.

Hi DzikuVx,

thx for your reply. I don't got you or you maybe not me :) The sensor is there and working in INAV, I just get this error. Can you maybe tell me what  I2C Errors: 5 means?

BR
D12A13

D12A13, 2023-06-02:
Here also a screenshot:
[image]

DzikuVx, 2023-06-11:
5 errors mean there were 5 errors on the I2C bus. I don't understand why you keep asking the same question

whwxxstsyd, 2023-06-15:
> ## Current Behavior
> First Thx for this great software and effort you are investing to make it avilable. INAV show I2C Errors: 5 on SPEEDYBEE F7 V3 and a BLITZ M10 GPS with compass module QMC5883L. The GPS Module and the mag compass are working fine, it just shows this error at the button line and at the status command. I get sensor output on the sensor tab. I tried I2C Speed 400 and 800 no change.
> 
> ## Steps to Reproduce
> 1. Set Magnetometer to QMC5883 on SPEEDYBEE F7 V3 and an BLITZ M10 GPS
> 
> ## Expected behavior
> I2C Errors: 0
> 
> ## Suggested solution(s)
> On Betaflight I was able to fix this with set mag_i2c_device = 1, but this command is not avilable at INAV
> 
> ## Additional context
> status INAV/SPEEDYBEEF7V3 6.1.0 May 11 2023 / 14:10:45 ([d6ce6e7](https://github.com/iNavFlight/inav/commit/d6ce6e7014ab33a97f13bafbfd25322ec3a08c5e)) GCC-10.2.1 20201103 (release) System Uptime: 262 seconds Current Time: 2023-06-01T17:09:34.349+00:00 Voltage: 0.00V (1S battery - NOT PRESENT) CPU Clock=216MHz, GYRO=BMI270, ACC=BMI270, BARO=BMP280, MAG=QMC5883 STM32 system clocks: SYSCLK = 216 MHz HCLK = 216 MHz PCLK1 = 54 MHz PCLK2 = 108 MHz Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK SD card: Manufacturer 0xc8, 491520kB, 05/2165, v12.8, 'APPSD' Filesystem: Ready Stack size: 6144, Stack address: 0x20010000, Heap available: 1716 I2C Errors: 5, config size: 9031, max available config: 16384 ADC channel usage: BATTERY : configured = ADC 1, used = ADC 1 RSSI : configured = ADC 2, used = none CURRENT : configured = ADC 3, used = ADC 3 AIRSPEED : configured = none, used = none System load: 19, cycle time: 515, PID rate: 1941, RX rate: 9, System rate: 9 Arming disabled flags: RX CLI OSD: MSP DisplayPort: BetaFlight Compatability mode (HD) [53 x 20] VTX: not detected
> 
> ## https://pastebin.com/9F4SnKnq
> * FC Board name and vendor:
>   SpeedyBee F7 V3 BL32 50A 30x30 Stack
> * INAV version string:
>   INAV/SPEEDYBEEF7V3 6.1.0 May 11 2023 / 14:10:45 ([d6ce6e7](https://github.com/iNavFlight/inav/commit/d6ce6e7014ab33a97f13bafbfd25322ec3a08c5e))
>   GCC-10.2.1 20201103 (release)

I got the same question,have you fixed it yet