iNavFlight/inav #9095: GPS over wifi with ESP8266

MahmoudHamandi, 2023-06-02:
We are attempting to send GPS signals over wifi. (Our GPS signals are computed from MOCAP system).
We are receiving the position data over wifi through the ESP8266 chip, and then sending the data to a T7motor FC running inav.
The ESP chip outputs MSP GPS messages over UART, connected to the UART3 pins of the FC.
However, INAV is not detection a GPS signal.
Inside INAV-configurator we have chosen UART3 as an MSP port, and chosen the protocol inside GPS as MSP.  
Please find below our ESP8266 code with dummy GPS data for testing: 
>#include <ESP8266WiFi.h>
#include <WiFiUdp.h>
#include <time.h>
#include <SoftwareSerial.h>
#include <Esp.h>
SoftwareSerial espSerial(3, 1); // RX, TX
void setup() {
  ESP.wdtEnable(0);
  Serial.begin(115200);
  espSerial.begin(115200);
  delay(1000);
}
void loop() {
  // Create and populate the GPS data structure
  struct __attribute__((packed)) {
    uint8_t instance;
    uint16_t gpsWeek;
    uint32_t msTOW;
    uint8_t fixType;
    uint8_t satellitesInView;
    uint16_t horizontalPosAccuracy;
    uint16_t verticalPosAccuracy;
    uint16_t horizontalVelAccuracy;
    uint16_t hdop;
    int32_t longitude;
    int32_t latitude;
    int32_t mslAltitude;
    int32_t nedVelNorth;
    int32_t nedVelEast;
    int32_t nedVelDown;
    uint16_t groundCourse;
    uint16_t trueYaw;
    uint16_t year;
    uint8_t month;
    uint8_t day;
    uint8_t hour;
    uint8_t min;
    uint8_t sec;
  } gpsData;
  // Populate the gpsData structure with fake GPS data values
  gpsData.instance = 1;
  gpsData.gpsWeek = 0xFFFF;
  gpsData.msTOW = 0;
  gpsData.fixType = 99;
  gpsData.satellitesInView = 99;
  gpsData.horizontalPosAccuracy = 10;
  gpsData.verticalPosAccuracy = 10;
  gpsData.horizontalVelAccuracy = 0;
  gpsData.hdop = 10;
  gpsData.longitude = 12345678;  // Replace with the desired longitude value
  gpsData.latitude = 87654321;   // Replace with the desired latitude value
  gpsData.mslAltitude = 0;
  gpsData.nedVelNorth = 0;
  gpsData.nedVelEast = 0;
  gpsData.nedVelDown = 0;
  gpsData.groundCourse = 15000;
  gpsData.trueYaw = 15000;
  gpsData.year = 2023;  // Replace with the current year
  gpsData.month = 5;    // Replace with the current month
  gpsData.day = 18;     // Replace with the current day
  gpsData.hour = 12;    // Replace with the current hour
  gpsData.min = 30;     // Replace with the current minute
  gpsData.sec = 0;      // Replace with the current second
  // Length of data and checksum calculation
  size_t len_data = sizeof(gpsData);
  size_t size = len_data + 6;
  uint8_t checksum = 0;
  // Create buffer
  uint8_t bufView[size];
  // Fill buffer with data
  bufView[0] = '$';
  bufView[1] = 'M';
  bufView[2] = '<';
  bufView[3] = len_data;
  bufView[4] = 201; // MSP_SET_RAW_GPS command
  checksum = bufView[3] ^ bufView[4];
  // Copy the gpsData structure into the buffer
  memcpy(&bufView[5], &gpsData, len_data);
  // Calculate the checksum
  for (size_t i = 0; i < len_data; i++) {
    checksum ^= bufView[i + 5];
  }
  bufView[size - 1] = checksum;
  // Send the message over the serial port (UART)
  for (size_t i = 0; i < size; i++) {
    espSerial.write(bufView[i]);
  }
  ESP.wdtFeed();
  delay(1000);
}

stronnag, 2023-06-02:
Perhaps you will have more luck if you feed the data that `MSP_SET_RAW_GPS` expects ....

MahmoudHamandi, 2023-06-02:
> 

Can you elaborate more?
I thought that what I used in the above code is the data that MSP_SET_RAW_GPS expects.

stronnag, 2023-06-02:
`src/main/fc/fc_msp.c` begs to differ

closed 2023-06-02 by DzikuVx

MahmoudHamandiNYUAD, 2023-07-12:
Eventually only sending the data requested in src/main/fc/fc_msp.c worked for MSP_SET_RAW_GPS.
Then we attempted to send MSP2_SENSOR_COMPASS, and we are out of luck.
Below is our code.

>typedef struct MspSensorCompassDataMessage {
    uint8_t instance;
    uint32_t timeMs;
    int16_t magX;
    int16_t magY;
    int16_t magZ;
}t_compass;
...
>void sendMSPSetCompass(t_compass currentcompass){
  uint8_t flag = 0;
  uint8_t compasschecksum = 0;
  uint16_t function = MSP2_SENSOR_COMPASS; % set to 0x1F04
  uint16_t payload_size = 11;
  espSerial.write('$');
  espSerial.write('X');
  espSerial.write('<');
  espSerial.write(flag);
  espSerial.write(function & 0xFF);
  espSerial.write((function & 0xFF00) >> 8);
  espSerial.write(payload_size & 0xFF);
  espSerial.write((payload_size & 0xFF00) >> 8);
  espSerial.write((uint8_t*)&currentcompass.instance, sizeof(currentcompass.instance));
  espSerial.write((uint8_t*)&currentcompass.timeMs, sizeof(currentcompass.timeMs));
  espSerial.write((uint8_t*)&currentcompass.magX, sizeof(currentcompass.magX));
  espSerial.write((uint8_t*)&currentcompass.magY, sizeof(currentcompass.magY));
  espSerial.write((uint8_t*)&currentcompass.magZ, sizeof(currentcompass.magZ));
>
 >compasschecksum ^= flag;
  uint8_t* ptrr = (uint8_t*)&function;
  for (size_t i = 0; i < sizeof(function); i++) {
    compasschecksum ^= ptrr[i];
  }
  ptrr = (uint8_t*)&payload_size;
  for (size_t i = 0; i < sizeof(payload_size); i++) {
    compasschecksum ^= ptrr[i];
  }
    ptrr = (uint8_t*)&currentcompass.instance;
  for (size_t i = 0; i < sizeof(currentcompass.instance); i++) {
    compasschecksum ^= ptrr[i];
  }
    ptrr = (uint8_t*)&currentcompass.timeMs;
  for (size_t i = 0; i < sizeof(currentcompass.timeMs); i++) {
    compasschecksum ^= ptrr[i];
  }
    ptrr = (uint8_t*)&currentcompass.magX;
  for (size_t i = 0; i < sizeof(currentcompass.magX); i++) {
    compasschecksum ^= ptrr[i];
  }
    ptrr = (uint8_t*)&currentcompass.magY;
  for (size_t i = 0; i < sizeof(currentcompass.magY); i++) {
    compasschecksum ^= ptrr[i];
  }
    ptrr = (uint8_t*)&currentcompass.magZ;
  for (size_t i = 0; i < sizeof(currentcompass.magZ); i++) {
    compasschecksum ^= ptrr[i];
  }
  espSerial.write(compasschecksum);
}