iNavFlight/inav #9098: Forward Feed Not Working On Yaw/Rudder With Plane With Tail

KenImhof, 2023-06-04:
Forward Feed on Stabilized Yaw has no impact in ACRO mode. The rudder travel end-points are also greatly reduced. When configured for 1000-usec low and 2000-usec high, the outputs are 1150 and 1850. Is there a CLI setting I am missing to get full travel as it is need on a foamy 3d plane for vertical hover. Roll and Pitch correctly output the full travel range (1000-2000usec) when Forward Feed is increased.

MrD-RC, 2023-06-04:
FF is relative to rates. So you will need to increase the yaw rate for FF to have a bigger effect.

KenImhof, 2023-06-05:
@MrD-RC thank you for your reply, but what do you mean by increasing the yaw rate.  I have an rx yaw min rate of 1000us and max rate of 2000us.  I also have the yaw rate per second set at 450 degrees per second and the FF on yaw has no affect even though it works correction on the Pitch and Roll.  Have you tried the FF on Yaw on your model and does it correctly increase the rudder/yaw servo output when in ACRO mode?

closed 2023-10-27 by DzikuVx