iNavFlight/inav #9099: Unable to use brushed reversible ESC for rover

a-loner, 2023-06-05:
## Current Behavior
I've tried STANDARD and BRUSHED ESC protocol, but they both failed to drive the motor. 
The servo works fine though. Specifically, the blue bar follows the stick movements on the radio, and the servo actually moves.

## Steps to Reproduce
1. The mixer tab is set as follow.
[image]
2. The outputs tab when trying STANDARD esc protocol  
[image]
   1. The **Motor Test** toggle grayed out and cannot be changed.
   2. The **Master** slider grayed out and cannot be adjusted
   3. The **Blue bar** doesn't follow the throttle stick movements on the radio
3. The outputs tab when trying BRUSHED esc protocol
[image]
   1. The **Reversible motor mode** option can be changed for the moment but always revert to **disabled** after "save and reboot"
   2. The **Motor Test** is automatically enabled and can be changed.
   3. The **Master slider** can be adjusted when **Motor Test** toggle is enabled.
   4. However, the motor makes no move.

## Expected behavior
I am not sure if it is necessary to meet the ARM condition when doing motor test, so I've made following attempts.
   1. Tried it outdoors with GPS fully initialized.
   2. Tried it with stick command to pass GPS check
   3. Tried it without connecting to PC/Configurator
   4. Tried different ESCs that behaves normally on testing equipment.
 
None of them works.  

## Suggested solution(s)
I know it seems weird, but I've tried connecting the servo output to motor ESC and it works fine, so I guess the FC is not generating the right signal or not outputing signal at all. I am not sure the problem is with my settings, my build or iNav (Configurator / Firmware).

## Additional context
The CLI dump is [here](https://pastebin.com/sLXVuGnu)

---
```bash
# version
# INAV/KAKUTEH7MINI 6.1.0 May 11 2023 / 12:37:54 (d6ce6e70)
# GCC-10.2.1 20201103 (release)
```

- FC Board name and vendor: Holybro Kakute H7 mini
- INAV version string: 6.1.0 May 11 2023 / 12:37:54 (d6ce6e70)

DzikuVx, 2023-10-27:
The brushed ESC is not the same as reversible ESC with brushed motor. By design BRUSHED protocol can not be reversible as it's directly driving the motor via a single MOSFET.
Reversible ESC has to be driven by `STANDARD` protocol

closed 2023-10-27 by DzikuVx

Babufrik8, 2024-11-02:
I have the Same Problem. I dont get a Command from Throtthe Stick, and I get no Signal on s1 for the esc. I tryed different escs but nothing works. I have the SpeedyBee 405 mini Wing. I only can make the esc work, when i choose non reversible Motors on Motor Test in the gui. But allready not from the ty