iNavFlight/inav #9102: Servo on multicopter Skystars H743

noobishe, 2023-06-05:
How to connect a servo on a multicopter with FC skystars h743? Where to connect the servo?
There are only 8 outputs for motors on the flight controller. 
S9 and S10 are specified in the target:

DEF_TIM(TIM15, CH1, PE5, TIM_USE_MC_SERVO | TIM_USE_FW_SERVO, 0, 0),   // S9
 DEF_TIM(TIM15, CH2, PE6, TIM_USE_MC_SERVO | TIM_USE_FW_SERVO, 0, 0),   // S10 DMA_NONE

mmosca, 2023-06-06:
You need to find where PE5 and PE6 are routed to.

They may not be exposed on the board, or may be shared with other features.

Some of the outputs not being exposed in pads is somewhat common when multiple fc share the same target.

DzikuVx, 2023-10-27:
will be fixed in INAV 7

closed 2023-10-27 by DzikuVx