iNavFlight/inav #9104: Geprc GEP-F405 HD flightcontroller Support

enderkruemel, 2023-06-06:
I have an geprc GEP-F405 HD flightcontroller, but there is no target in inav that i can use to flash it.

YI-BOYANG, 2023-06-08:
[image]
Hello, the GEPRC team is working on the adaptation and plans to support it in the next version update

enderkruemel, 2023-06-08:
Im very Happy to hear, that they working on the Implementation. I was thinking that i have to mode an custom target on my own... 
Thanks for the info
:-)

enderkruemel, 2023-07-15:
the fc target is still not aviable 6.1.1...

enderkruemel, 2023-09-09:
i tryed it now it works, but the sensor alingment of the accelerometer is 90* rotated. also seems like the build in barometer isnt detected in inav. its working in betablight.  so the target needs a few adjustments.

YI-BOYANG, 2023-09-12:
> Hello, the accelerometer Angle and barometer Settings are modified based on the information in the BF.

enderkruemel, 2023-09-12:
If i use betaflight the gyro has the normal orientation, in inav i have to rotate it 90* but that is easy to fix. 
But the Barometer doesnt show up and if i set the right type manualy it only makes the copter beeping ant the icon on the top is red. I also tryed to have a battery plugged in , but this also doesnt seem to work.  For me this Looks like an Problem with the target...

tchleb, 2023-10-21:
I found the [fault](https://github.com/iNavFlight/inav/pull/9382/commits/d2b134a17654584328d2bcca4cfda9e9462f202b). 
There is a fault inside the iNav / BF target. Internal BMP280 is connected to I²C_2 (SCL PB10; SDA PB11). 
In the target this is on i2c_1.

I created a [PR](https://github.com/iNavFlight/inav/pull/9382).

enderkruemel, 2023-10-25:
Ok thank you. I2c2 works for me now. 
Its because there are2 Versions of the Flightcontrollers.
I have the original one, and also the betaflight target uses this older Board. 
but the inav target is based on the v2 of this fc. 

Its very confusing if bouth Flightcontrollers use the same target, and have different Hardware configurations

But Switching the i2c port works. only the gyro is rotated 90° but thats easy fix.

closed 2023-10-25 by enderkruemel

YI-BOYANG, 2023-10-25:
Can you compile the firmware yourself? If not, I can provide you with the firmware modified to match your hardware.

enderkruemel, 2023-10-25:
I never tryed it... It would be very nice if you can send me the fixed Version

tchleb, 2023-10-25:
> I never tryed it... It would be very nice if you can send me the fixed Version

[inav_6.1.1_GEPRCF405_BMP280_I2C-2.zip](https://github.com/iNavFlight/inav/files/13163147/inav_6.1.1_GEPRCF405_BMP280_I2C-2.zip)

It will work with this [FC](https://github.com/iNavFlight/inav/pull/9382#issuecomment-1773794427)

YI-BOYANG, 2023-10-25:
[inav_6.1.0_GEPRCF405(i2c2).zip](https://github.com/iNavFlight/inav/files/13163238/inav_6.1.0_GEPRCF405.i2c2.zip)
This is version 6.1.0, if you need it.

tchleb, 2023-10-25:
> Ok thank you. I2c2 works for me now. Its because there are2 Versions of the Flightcontrollers. I have the original one, and also the betaflight target uses this older Board. but the inav target is [...]

Yes it is confusing that GEPRC use the same target with differnt pinout. I guess they want to milk there cashcow and hope that the user will buy there M10 GPS with include Kompass + Baro. (Currently no others sell a GPS with incl. Baro)

@DzikuVx what is your opinion for creating a second target? Do you think it will be confusing the iNav users?

YI-BOYANG, 2023-10-25:
Brother, your conjecture is groundless.
I will give feedback to the GEPRC team on the issues discussed today. Thanks.

tchleb, 2023-10-25:
> Brother, your conjecture is groundless.
> I will give feedback to the GEPRC team on the issues discussed today. Thanks.

Why groundless?

For my the only logic behind is that GEPRC saved the cost for BMP280 (~1$). On the PCB layout is enough space available for it. The I2C_2 pins are also not used for other things.

YI-BOYANG, 2023-10-25:
If you look at other FC in GEPRC, you will see that in many FC products launched in the past, there was no barometer integrated either, long before GPS integrated barometer was introduced, not just now.

Because before this, GEPRC's FC was mainly designed according to the use scenario of Betaflight, the integrated barometer was not a necessary option, and the cost saved would also be more preferential in product pricing.

The introduction of GPS integrated with barometer is to consider the need of some FC for barometer (not only GEPRC), because many FC only integrate BMP280 or even do not have barometer, and GEPRC GPS can provide better DPS310 or MS5611, of course, the price will be slightly more expensive. But the purchase option remains in the hands of the user.

Thank you very much for your suggestions. I will feed back to the R&D team of GEPRC for discussion. Finally, I would like to introduce GEPRC's GPS integrated with barometer to you. If you want to try it out, please send me the shipping address so that I can send you some tests.

tchleb, 2023-10-26:
Yes i agree in paste not all FC have barometer integrated. In Betaflight the Barometer is not really needed. 

I think it's good that GEPRC has brought a GPS onto the market that has an integrated compass and a barometer. That didn't exist yet. Until now, the only option was to retrofit the Baro using an external PCB.

I'm interested in how the baro behaves when it gets more wind then inside the stack. 

I sent you an email with my address to your github mail.

YI-BOYANG, 2023-10-26:
I can't seem to find the email you sent me, if you can, please resend it to: yangboyi@geprc.com. Or reach out to me on discord: yi_bo_yang.

govt20job-ai, 2025-08-16:
Barometer is not available in geprc_F405-HD fc