iNavFlight/inav #9112: Servos get commanded past configured endpoints upon reboot via configurator

malinhiles, 2023-06-10:
## Current Behavior
I have a flying wing aircraft setup with a traditional elevon mix. When changing saving and rebooting settings in the configurator, if the servos are powered, they get promptly commanded beyond their set endpoints and cause damage to my aircraft. This happens more than half of the time when saving and rebooting. 

## Steps to Reproduce
1. Power flight controller and servos
2. Connect to configurator
3. click save and reboot on any of the configurator tabs that have it
4. observe 🙌

## Expected behavior
I'm sure it may be apparent given the nature of the issue, but for consistency I'll fill this section. The servos shouldn't suddenly move to their absolute extremes after save and reboot. They shouldn't move at all.

## Additional context
I'm using digital KST servos running at 333Hz

https://pastebin.com/HDQeZWtb

Matek H743 - WLITE

# version
# INAV/MATEKH743 6.0.0 Mar 22 2023 / 10:48:04 (084101ac)
# GCC-10.2.1 20201103 (release)

b14ckyy, 2024-03-21:
This servo extreme movement is an issue for years now. The problem is, that this is not something that INAV can solve. The extreme positions are caused by erratic IO states during the boot process of the MCU before INAV is even fully initialized and enables the outputs itself. 

I don't know if this is a design flaw of the MCUs, the Flight controller or the servo controllers but it is nothing INAV can fix. Sorry.

closed 2024-03-21 by b14ckyy

sensei-hacker, 2024-03-21:
FYI there is a recent change that *may* help in the next version.  INAV can't guarantee what the servos will do when INAV isn't running, but it might help.