iNavFlight/inav #9117: Servo mixes using RC throttle as input do not work properly.

malinhiles, 2023-06-11:
## Current Behavior
Servo mixes using RC throttle as input do not work properly. Here is a basic flying wing example using a fresh and clean flash of 6.1.0:

If I setup a mix with one of the servos and the rudder channel, everything works as expected and the mix actuates the servo properly.
[image]
[image]

When I change the input for that mix to RC Throttle (or even stabilized throttle), and make no other changes, the outputs tab shows (and actually outputs) the servo pwm pinned to the limit. No matter how I change the value of the throttle channel coming in via crsf, the mix remains pinned at the max PWM.
[image]
[image]

## Steps to Reproduce
flash 6.1.0, and try both of the scenarios listed above. 
## Expected behavior

## Additional context
this was not a problem in 6.0.0
https://pastebin.com/qbQgS7Gw)
---
# version
# INAV/MATEKH743 6.1.0 May 11 2023 / 13:29:16 (d6ce6e70)
# GCC-10.2.1 20201103 (release)

- FC Board name and vendor: MATEK H743-WLITE

MrD-RC, 2023-06-11:
Have you armed? The throttle won’t move unless you arm. It is forced to LOW.

malinhiles, 2023-06-11:
This is without arming. Are basing this on the recent change in 6.1 where the "Stabilized Throttle" value now only moves when armed? (https://github.com/iNavFlight/inav/issues/8889)

I am using "RC Throttle" as input for the mix instead of "Stabilized Throttle" for that reason, however, this issue is apparent with both regardless.

MrD-RC, 2023-06-11:
Actually, that change requires arming for both Stabilised and RC throttle for safety. Arming is a necessary step in INAV.

malinhiles, 2023-06-11:
Hmm, interesting. Why is the mix forcing the servo value to HIGH then? The weight of the mix is only 10%. 

And what is the safety aspect here? The output of this mix is mapped to a servo channel, not a motor channel. It seems a bit clunky to have the servos maxed out when the plane is just sitting on the ground disarmed because of this mix.

MrD-RC, 2023-06-11:
This is an unrealistic mixer. You wouldn’t have your elevons on the same servo output as the throttle. That makes no sense in INAV at all. So there would be no chance of disarming pulling the elevon servo low on disarm. If the mixer is set up correctly.

malinhiles, 2023-06-11:
Respectfully, I disagree. I used the following mix in 6.0.0 to reduce pitch effects when at different throttle values. It's a mix with a 2% weight that adds slight adjustments to my elevons as I fly at different speeds. 
[image]

Its the very same mix that us wing racers make in our radios when we are flying a racing wing that doesn't have an FC.
And the same mix that pylon fliers make with their rudder and aileron to counteract motor effects when their mount doesn't have the right offset angle.
And its the same mix that us hobbyists make when an aircraft we buy wasn't designed with the right thrust angle.

yes, i-gain can be used to address a similar issue, but the PIDs aren't making this correction in manual mode. 

If you have a better way to implement this mix that avoids this issue, I'm open to considering it.

MrD-RC, 2023-06-12:
This isn't ArduPilot. Thankfully, we don't do things the same way they do. Besides, ArduPilot doesn't even have this mixing capacity. There is the TECS to adjust the throttle based on how the aircraft is flying. But I've not seen a way to change the mix based on throttle at all. Neither have I seen any separation of RC throttle or stabilised throttle. 

The only difference is the arming. With AP, if you're disarmed. The throttle will still be registered. It just won't output anything. Perhaps this would be a better approach. 

But I still think a mixer like that is unnecessary. Unless you're flying in manual. But even then; surely it would be better to correct the thrust angle. So that you don't have these issues in the first place.

DzikuVx, 2023-06-12:
Like @MrD-RC stated, this is not a bug. This is intended behavior introduced to solve a safety issue of motors spinning without pilots intention.

I do get that there are some extremely rare cases when THR can be mixed to a control surface. They are not common.

If @malinhiles you would like to have THR mix that works even when disarmed, it's doable via Programming Framework.

closed 2023-06-12 by DzikuVx

malinhiles, 2023-06-19:
> Like @MrD-RC stated, this is not a bug. 
Understood. Just changed the title of the thread. 

>This is intended behavior introduced to solve a safety issue of motors spinning without pilots intention.
The mix is just using the throttle channel as an input not as an output. A servo channel is the output. Even so, the throttle channel PWM output is held LOW until arm.

> If @malinhiles you would like to have THR mix that works even when disarmed, it's doable via Programming Framework.
Thanks for the suggestion, I'll check this out. 👍