iNavFlight/inav #9119: Mamba F722 mini AIO can't arm when DSHOT300 or 600 enabled

DiscoMan18, 2023-06-11:
When I set DSHOT300 or 600, the quad doesn't arm with the message ARMING_DISABLED_ARM_SWITCH. When I set standard or multishot, it arms normally. Tried BLHELI_S and Bluejay firmwares, nothing changes.

mmosca, 2023-06-13:
Which target?

DiscoMan18, 2023-06-14:
> Which target?

Mamba F722 I2C

mmosca, 2023-06-14:
I have a diatone f722 mk1 aio and it works fine with dshot600. 

I can’t find an exact match for your fc in their website but the newer versions seem to use a different target. 

MAMBAF722_2022A or MAMBAF722_2022B

you could try that as well.

DiscoMan18, 2023-06-24:
I'm closing the issue because everything works now. Nothing really changed in the settings or setup. Seems to not be a bug.

closed 2023-06-24 by DiscoMan18

DaniilProgMath, 2024-03-28:
Same problem, dshot300 and dshot600.
armed only occasionally. The error flag is the same.
All other arm locks are successful passed.

Flight Stack speedybee f405 bls 55 A.

these protocols work in betaflight.

mmosca, 2024-03-29:
> Same problem, dshot300 and dshot600. armed only occasionally. The error flag is the same. All other arm locks are successful passed.
> 
> Flight Stack speedybee f405 bls 55 A.
> 
> these protocols work in betaflight.

Posting on a close issue that has been confirmed working is probably not going to help you. 

Try dropping in on discord for help or opening a new issue with more information

DaniilProgMath, 2024-03-29:
> Posting on a close issue that has been confirmed working is probably not going to help you.

I understand that it was more of a cry from the heart. I didn’t think that anyone would hear me at all, thanks for the advice. I’ll open a new issue, perhaps since there have already been two cases with an error, it means there will be more.