iNavFlight/inav #9129: Programmtically set osd_hud_homepoint = ON/OFF

sdellava, 2023-06-16:
## Current Behavior
osd_hud_homepoint can be set ON/OFF only in CLI

## Desired Behavior
Allow to set programmtically the osd_hud_homepoint = ON/OFF 

## Suggested Solution
Add the osd_hud_homepoint = ON/OFF  in the list of action that can be taken when a logic condition is true.

## Who does this impact? Who is this for?
Every INAV pilots that use osd_hud_homepoint

## Additional context
osd_hud_homepoint is a useful function when the aircraft is far from the homepoint, but due to the low resolution of the OSD, it is annoying and disturbing when the aircraft is approaching the homepoint, such as for landing. In this case, the pilot can set osd_hud_homepoint = OFF using an AUX channel or it can be programmatically set to on or off based on the distance to the homepoint.

jc9aj, 2023-06-16:
This feature can already be achieved by switching between OSD layouts. setup one with the home arrow and one without. You can then switch between them with a manual switch. 

In addition, you can setup INAV to automatically switch the OSD layouts using the programming framework based on the distance to home (or other suitable parameter)

sdellava, 2023-06-16:
I'm not referring to the home arrow in the OSD, but the hud or radar configuration that, eventually, shows the home point mapped in the OSD display area, exactly like other planes sending their position with the onboard ESP32 beacon

rts18, 2023-06-17:
@sdellava You forgot to add that it requires the crosshair to be enabled for all of these to work.
  `osd_hud_homepoint = ON` 
  `osd_hud_homing = ON`
  `osd_hud_wp_disp = 0 - 3`

#8838

sdellava, 2023-06-18:
Yes, off corse.

Maybe a good and simple solution is to allow to define which OSD profiles
enable the Hud feature. So if I want to turn off the hud I can simply
switch the OSD  profile to the one that has the hud off.

This could be a useful feature also for those who use the hud to view
others aircraft: to have a better view a specific OSD layout could be
preferred.

Jetrell, 2023-06-18:
I like the idea... But I think it still limits versatility for all the HUD features.
As stated above.. The Crosshair has to be selected on a specific alternate OSD page, to display any of the HUD's..

But if the user has `osd_hud_homepoint`,  `osd_hud_homing`, `osd_hud_wp_disp` and the `ESP32 radar HUD` **all** turned ON.. Then what ever OSD Alt page the Crosshair is selected, will display them all.. Which makes the screen a total mess of characters clashing all over the place.

I think a better idea. Would be to have the 4 HUD features selectable in the _OSD Maps and Radars tab_.. Being independent of the Crosshair... This way the user can place one HUD feature on _OSD Alt 1_ and another on _OSD Alt 2_ and another on _OSD Alt 3_. Or any combination.

[image]

mmosca, 2023-07-28:
@sdellava @rts18
Testing help for #9199 is appreciated. I haven't used the hud features much before, so I am not used with how they should work.

If you could test it and provide feedback, it would be greatly appreciated and make it more likely to get the changes merged.

mmosca, 2023-08-06:
> @mmosca I've cloned and compiled your branch (for target FLYWOOF405NANO that actually has never been added as official target). I hope to test it tomorrow.
> 
> UPDATE: I can't test beacuse for some reason the GPS fails and without a configurator I'm not able to debug the issue.

The update didn't ping me, so I missed it.

I believe I just fixed the bug that was preventing your gps from being detected, so you should be able to try now.