iNavFlight/inav #9133: NO RC LINK occurs under multiple circumstances

thujones, 2023-06-18:
**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****
Diatone Roma F4 LR Frame
HolyBro H7 Mini FC
HolyBro Tekko32 F4 4in1 min 45A ESC
UART2 - Rush Tank Mini VTX (SmartAudio)
UART3 - Flywoo Goku GM10 Mini v3 (Compass wired to SDA/SCL)
UART4 - RunCam Camera Control (Runcam Phoenix 2 JB Edition)
UART6 - BetaFPV ELRS Nano Receiver 2.4GHz (Tx is BetaFPV ELRS 1W version)
UART7 - ESC Telemetry
Standard buzzer on buzzer pads
3000mah LiON battery

## Current Behavior
While performing ground based actions such as OSD settings saves, waiting for GPS to lock, often receive NO RC LINK. Prior to 6.x I do not recall this issue occurring really at all.

Have reflashed with Betaflight, issue does not occur. Originally attempted remediation by changing from Namimno RC TX and RX to BetaFPV TX and RX and issue still occurs.

One thing to note is that during regular flight, even "long distance" flights (Example recent flight https://www.youtube.com/watch?v=I8W0HjkqAjY ) It works no problem, link is practically 100% stable and over dozens of flights have never had a mid-air NO RC LINK event, indicating that potentially issue revolves around various subsystem initialization or saving routines

About this time last year on v4.10 and up until recently (Example flight https://www.youtube.com/watch?v=WtvEjL9cEoA ) I cannot recall this ever happening

## Steps to Reproduce
Note: There appears to be multiple ways to instantiate this one, most common I can find are lined out below:

1. Start quad up cold, can be via battery or from USB (although typically now I use battery to monitor the video output so I know when this occurs)
2. Be on iNAV Configurator receiver tab, inputs from Radio are accepted and shown as expected
3. When issue occurs, inputs are shown on channel map, work fine
4. Quad may beep a moment later (not sure what was initializing or reinitialized) its at this point the rc input no longer works, and NO RC LINK is shown
4a. This can also be induced pretty much on demand by using the OSD stick inputs, making practically any change and saving

## Expected behavior
RC Link to be read by FC and stable throughout use of model when radio is obviously within range

## Suggested solution(s)
Not sure, it seems to have started around the release of 6.x series

## Additional context
While I am currently using a custom build that forces GPS rate to 10Hz by way of direct edit of all set rate functions in ublox, issue occurs with stock version of 6.x firmwares as well.

https://pastebin.com/EQpKniTA

---

- FC Board name and vendor:  HOLYBRO KAKUTE H7 MINI
- INAV version string: 
-
# version
# INAV/KAKUTEH7MINI 6.1.1 Jun 17 2023 / 13:01:11 (42ced253)
# GCC-10.2.1 20201103 (release)

b14ckyy, 2024-03-22:
does this still accurr with latest 7.0 or 7.1RC1 releases? Also latest ELRS Firmware?

b14ckyy, 2026-09-22:
inactive

closed 2026-09-22 by b14ckyy