iNavFlight/inav #9134: Inav main (7.0.0) repo could not build on mac (M1 proc) for MATEK F411WTE board

knoppixmeister, 2023-06-18:
## Current Behavior

Inav main (7.0.0) repo could not build on mac (M1 proc) for MATEK F411WTE board.
Some problem with linker (LD)

[image]

## Steps to Reproduce

1. clone inav repo
2. cd to cloned dir
3. mkdir build && cd build
4. cmake ..
5. make -j4 MATEKF411TE

- FC Board name and vendor: MATEK F411WTE
- INAV version string: inav_7.0.0

stronnag, 2023-06-19:
Works fine (Linux)
```
$ cd build
[jrh@eeyore]$ ninja MATEKF411TE
[355/356] Linking C executable bin/MATEKF411TE.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:         776 B        16 KB      4.74%
    FLASH_CONFIG:          0 GB        16 KB      0.00%
          FLASH1:      406721 B       480 KB     82.75%
             RAM:       84108 B       128 KB     64.17%
       MEMORY_B1:          0 GB         0 GB
[356/356] cd /home/jrh/Projects/fc/ina.../inav/build/inav_7.0.0_MATEKF411TE.hex
```

Possibly some collateral damage from the MacOS SITL work (@mmosca) ?

stronnag, 2023-06-19:
The problem is that LTO is disabled if the host is "APPLE", I presume for SITL. This is suboptimal for firmware on F411/F722 in particular.

cmake/main.cmake
```
 if(IS_RELEASE_BUILD AND NOT CMAKE_HOST_APPLE)
        set_target_properties(${exe} PROPERTIES
            INTERPROCEDURAL_OPTIMIZATION ON
        )
    endif()
```

PR #9136 (2023-06-19): reenable LTO for MacOS firmware builds
Fixes the firmware build regression from #9082, retaining no-lto for MacOS SITL builds. Fixes #9134
closed 2023-06-19 by stronnag