iNavFlight/inav #9135: OSD and Lua script doesn't agree about the current heading

Rodrigo-NH, 2023-06-19:
## Current Behavior
OSD and Lua script doesn't agree about the current heading
[image]

## Steps to Reproduce

1. Calibrate accelerometer and compass
2. Wait for GPS fix and ready to arm
3. Compare values at different angles (N, E, S, W, etc)

Interesting part is OSD and LUA agrees for some angles but not for others. Repeated this test a dozen times (with some reboots in between), and the general output was:

Direction | OSD reading | Lua reading
N | 345 | 2
E | 73 | 73
S | 157 | 156
W | 285 | 302

In this test batch all the readings followed this pattern where it agrees about E and S but not about N and W

## Expected behavior

Information in OSD and Lua script to agree

## Suggested solution(s)

I'm not sure

## Additional context
- Using Garmin etrex32 built in compass as initial reference to compare accuracy (but not need this to test for this issue)
- Using Garmin reference it looks like iNAV OSD will show magnetic heading until it gets a GPS FIX, then correct to True North after GPS fix. (Didn't test for this from setup tab or Lua scripts but OSD)
- I see other oddities like iNAV and garmin agreeing completely sometimes about North, East and West but not South (with readings around 160º e.g.). Saw this same thing happening in other directions (like agreeing about North, West, South but not East).
- Making tests over a flat surface also different flat surfaces (other tables e.g.). The compass test looks right from setup page (making sense generally and no relevant deviation from current heading at different pitch angles)
- In my region the magnetic deviation is as huge as -23º so perhaps I'm in a better situation to observe details in this regard in iNAV.

Dump: https://pastebin.com/upaVHUmE

# version
INAV/SPEEDYBEEF405V3 6.1.1 Jun 13 2023 / 12:27:32 (42ced253)
GCC-10.2.1 20201103 (release)

thujones, 2023-06-19:
I noticed this yesterday as well. Seems at least in my case the telemetry isnt reporting Hdg anymore (always 0.00 degrees) and I assume the only other way that the reported compass heading can be calculated is from the Yaw telemetry which is reported in Radians. I dont think that in itself is a problem but it does certainly seem like some sort of weird scaling issue.

Rodrigo-NH, 2023-06-19:
My current problem is iNAV needs compass calibrated very well to be able to do nav modes but I don't know which value (OSD or Lua) to trust while checking calibration also the inconsistency between N->E->S->W angle difference (expected 90 but I get 15º ~ 23º deviations).

Did other tests right now and:
OSD and Lua agrees about heading for N and E before a GPS fix (S, W off by ~15°).
It follows the previous pattern I reported after GPS fix.

Interesting thing is Lua totally agreed with Garmin's compass for true north (off by 1º) before GPS fix in these last tests. (Didnt check this before)

b14ckyy, 2023-07-13:
Afaik the INAV OSD Compass shows the true heading of the plane based on Compass. The telemetry sends the course over ground to the LUA. 
before GPS fix both use raw compass data with magnetic declination taken into account for true north but after GPS fix, the position jumps that might occurr can then deviate the ground course data from the AHRS. This causes discrepancies after the GPS fix and is totally normal. 
as the Gyro is the primary 3D Attitude source and compass is only used for drift compensation over time, the estimated ground course during the sat log will stick until you go and fly. 

after that you will have ground course on the lua and heading on the OSD.

Rodrigo-NH, 2023-07-13:
Thanks for explanation!

Checking correct heading before flying is in the "best practices" list. So as I understand the Lua script is useless in this regard.
Perhaps a good ideia making it clear what heading really means as it actually means different things, in different stages/sources.

closed 2023-07-13 by Rodrigo-NH