iNavFlight/inav #9141: FRSKY ACCESS 24 Channel SBUS Support

chucky-whiskers, 2023-06-26:
## Current Behavior

Only 18 channels available in INAV, and limited to 16 channels via SBUS
## Desired Behavior

I want to be able to use 24 channel mode for my TD MX receivers
## Suggested Solution

Increase maximum amount of channels in INAV to 24, and add support for FRSKY ACCESS 24-channel SBUS mode

## Who does this impact? Who is this for?

Anyone who uses ETHOS transmitters, or ACCESS transmitters, or TD transmitters
## Additional context

DzikuVx, 2023-06-28:
SBUS protocol does not support 24 channels. It supports 16 full range and 2 on/off channels only

closed 2023-06-28 by DzikuVx

MrD-RC, 2023-06-28:
SBUS supports 24 channels on ACCESS. Though it seems to work differently to the 16 channel implementation. I did try to get this working. But no luck so far. The Jeti implementation of 24 channels works fine though. 

https://github.com/iNavFlight/inav/tree/MrD_Increase-channels-to-24-Not-for-F411-or-F722

frobert0, 2024-06-12:
> SBUS unterstützt 24 Kanäle auf ACCESS. Allerdings scheint es anders zu funktionieren als die 16-Kanal-Implementierung. Ich habe versucht, es zum Laufen zu bringen. Aber bisher ohne Erfolg. Die Jeti-Implementierung mit 24 Kanälen funktioniert jedoch einwandfrei.
> 
> https://github.com/iNavFlight/inav/tree/MrD_Increase-channels-to-24-Not-for-F411-or-F722

SBUS unterstützt 24 Kanäle auf ACCESS 
Würde auch gerne 24 Kanäle Modus haben, bräuchte 17 über SBus