iNavFlight/inav #9144: Mamba F740 mk4

stavross118, 2023-06-28:
Hi hope someone can help
I have the above FC on target mamba F722_2022b
Issues are the barometer worked on 6.0.0 now since this
It no longer detects it ....it was under msp tried them all 
Nothing works...
The speedy bee app as it's WiFi on board has never worked previously on here advised to remove sd card 
It doesn't have one ....
Really want to try INAV is there any aio board speedybee ect that will work with baro ect... out of the box and flashed or do I have to learn how to write or re write the firmware any help much apreciated

stavross118, 2023-07-03:
Using INAV 6.0.0 baro working also needed a servo so managed to use the led_strip
B03 pin as timer and removed the led strip ...and re built hex awaiting test on servo leg Incase anyone needs it its for INAV  MAMBAF722_2022B will upload it if anyone wants to check it or needs it ?

b14ckyy, 2024-03-23:
This issue seems to got missed. 
Please check with 7.0 or 7.1 RC1 if the issue still appears.

closed 2024-04-08 by b14ckyy