iNavFlight/inav #9150: RUSH_BLADE_F7 BMP280 Barometer doesn't work (need Fix)

NoviKurniawan, 2023-06-30:
[image]
NEED FIX
Current Behavior
RUSHFPV RUSH BLADE F722 FLIGHT CONTROLLER FOR DJI DIGITAL FPV SYSTEM supports BMP280 (and it works fine in Betaflight and Emuflight), but not in iNav. 
I tried the latest update (INAV 6.1.1) but still got the same error
thank you.

TeoVal138, 2023-07-21:
I have the same baro problem on my FLYWOO F745 : (

NoviKurniawan, 2023-08-16:
this case tested in Inav 7.0.0? INAV 6.1.x still error.
when is INAV 7.0 released?

madhukar01, 2023-12-15:
> <img alt="image" width="869" src="https://private-user-images.githubusercontent.com/966811/257457181-ae3e4ed4-df31-403b-857c-b4d664783f4c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI2NjIyNjEsIm5iZiI6MTcwMjY2MTk2MSwicGF0aCI6Ii85NjY4MTEvMjU3NDU3MTgxLWFlM2U0ZWQ0LWRmMzEtNDAzYi04NTdjLWI0ZDY2NDc4M2Y0Yy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBSVdOSllBWDRDU1ZFSDUzQSUyRjIwMjMxMjE1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDIzMTIxNVQxNzM5MjFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hMDY2MmQ3ODc5OTllZGRhOWQzMmNlMDIyNTQ3YzY4YTZhZDAwMTMxYTRiZmFlNThlZTJiYTA0OGIyMTkzODY5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.ROWECY3uxQ3_oC2tb0A_s385sI1syDy02sZ_gqSJt1s"> Just tested with Rush Blade Analog and it's working fine over here

This is the newer version of the RUSH BLADE with ICM Gyro.
The older version with MPU6000 still has the issue - checked today with INAV 7.0.0