iNavFlight/inav #9152: INAV 6.1.1 Rth Mode missing

boomerschnecke, 2023-06-30:
Rth Mode is missing with INAV 6.1.1
Or is there a different mode with same feature existing?

**PLEASE MAKE SURE YOU READ AND UNDERSTAND THE SOCIAL MEDIA SUPPORT CHANNELS. QUESTIONS ABOUT FLASHING, CONFIGURING, PILOTING MAY BE CLOSED WITHOUT FURTHER INTERACTION.**

* [INAV Discord Server](https://discord.gg/peg2hhbYwN)
* [INAV Official on Facebook](https://www.facebook.com/groups/INAVOfficial)
* [INAV Official on Telegram](https://t.me/INAVFlight)

**Please double-check that nobody reported the issue before by using search in this bug tracker.**

**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****

## Current Behavior

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

MrD-RC, 2023-06-30:
RTH is not missing. It won’t appear until you have a GPS setup.

closed 2023-06-30 by MrD-RC

boomerschnecke, 2023-06-30:
Hello Daren, GPS is ACTIV ! This is the reason why I wrote this issue. I have  12 satellites. So it should work normaly, but its not

boomerschnecke, 2023-06-30:
The copter is flying, but not Rth

MrD-RC, 2023-06-30:
Is your mag set up? RTH also needs mag with a multirotor.

boomerschnecke, 2023-07-02:
Hello Darren, thanks for feedback.  The new GNSS with mag is ordered.
In the meanwhile I tried to install the Matek Lidar System to the SpeedyBee F405V3
But unfortunatelly its not working. It shows two red symbols Flow and Sonar.
I did connection to the UART3 with MSP, and within the Configuration I did implement Rangefinder = VL53L0X and for Optical flow = MSP
But this did not work 
Could you please give me some hints whats wrong in my setup?
Thanks in advance

boomerschnecke, 2023-07-03:
Hi Darren, Hi Pawel,
sorry for disturbing, but I have some problems/ questions to the copter setup.

I did setup with SpeedyBee F405V3 with the ESC board, Jeti Receiver. 
So the copter is flying, but in the take off phase, there I have problems.  It happens very often, that the copter is immediatelly turning around and making a kind of chicken dance till disarmed.
Therefore I selectet "turn off motors at low throttle", in order to stop this immediatelly. But this is not working. So I could only stop with disarming switch.
While waiting for the ordered GNSS with mag, I installed the Matek Lidar Modul over the weekend.
I did the calibration sequence(10,5), and the first flight was ok even, the Lidar seems not to work.
I did a recalibration of the Lidar (9,5).
But then, I was not able any more to take off. Only chicken dance.
After deinstallation of Lidar, the copter was flying again.

So my questions:
1) for take off: is there a possiblity to adust the system for smooth take off?
I think normally the copter should take off rapidly from the ground like a heli with flybarless system?>
But with all this chicken danke at the gnd....
2) how do I get  "turn off motor at low throttle" to work? (Till system works fine)
3) Lidar: the arrow on the lidar, how should be the direction? same direction as FC (my one is turned by 180°) or to the front?
Are there any special things I should know about? Is for the Lidar also the Mag necessary?
But why was the first setup with Lidar flying, and the recalibration setup not flyable?
Thanks for you support in advance

boomerschnecke, 2023-07-03:
on the flying area I did some adustments concerning Lidar calibration via the SpeedyBee Bluetooth tool. Could there any negativ influence? Is "save and reboot" ok, or do I have to repower the copter again via Bat disconnection and reconnection?