iNavFlight/inav #9153: Inav 7.0.0 last build not working on SPEEDYBEEF405V3

alccrl, 2023-07-01:
HI,

trying to apply the (complex) fix for sd card spi known problem with OSD on SpeedyBee target SPEEDYBEEF405V3 6.1.1,

I compiled  yesterday inav-master 7.0.0 for SPEEDYBEEF405V3.

NEW PROBLEM:
Gps on UART 6 does not work on 7.0.0 SPEEDYBEEF405V3.

1)UART 6 does not show any UBLOX input (quad is the same and connections are the same).

This happen either modifying default conf 7.0.0 (starting from scratch flash) or using 6.1.1 conf dump (some red lines on gps stuff appear).

2)Moreover,  UART4 (BlueTooth (?)) causes IMU to be crazy if activated for MSP (this happens also on 6.1.1).

3)Baro is not recognized (6.1.1 shows SPL06, 7.0.0 show none)

## Expected behavior
These things should not happen with SPEEDYBEEF405V3 (consolidated target).
This is my last interaction with this sw, I lost too much time in trying to put things to work,
it is an endless effort.

Very sad ...

Alfredo

shirase, 2023-07-05:
Try to insert sd card. Bug is reproduced only without an SD card.

DzikuVx, 2023-07-05:
@shirase do you know what is the reason already? And of course, do you think you are able to help with a fix?

shirase, 2023-07-05:
@DzikuVx Maybe. Now i doing some tests.

alccrl, 2023-07-05:
Osd gets wrong when inserting an sd card after power on, or you power on wth an sd card.
I am testig with arducopter (elrs included in rx tx system) and sd card seems ok, 
I have yet to fully configure and fly.
Got problems with edgetx and lua scripts, but this is another story :-)

Alfredo

DzikuVx, 2023-07-05:
thank you very much @shirase

shirase, 2023-07-05:
@alccrl can you try my fix https://github.com/shirase/inav/tree/9153-sdcard-osd

alccrl, 2023-07-05:
Thanks,
I will, but I have to "queue" this activity because at the moment I have yet to finish test of arducopter installation. Then it is necessary to save arducopter conf and to flash inav again. It will take me time, I live in town and I cant fly with untested sw in crowded places ...:-).

I'll let You know.

Alfredo

alccrl, 2023-07-07:
I made a partial test and osd is ok, but I could not end test because gps on your build does not work, uart6 does not respond.
So I had to go back to 6.1.1 with no sd card.
In the meantime I broke usb connector (it detached from fc) so I have much bigger troubles....I'll try to refix it someway (uart4 bt is not enabled, I disabled it rying to put to work gps)...
Moreover arducopter does not handle correctly elrs on crsf and irc protocol does not work, so I am very much confused .....:-)
Alfredo

PR #9162 (2023-07-12): Fix: Inav 7.0.0 last build not working on SPEEDYBEEF405V3
Bug appears if you use analog OSD on SPEEDYBEEF405V3 and do not insert sd card.
SD card try to initialise many times and eche time slow down SPI bus speed, but analog OSD works on same SPI bus. When spi slow down, analog OSD slow and increase pid loop. It make unstabile gyro calibration. That fix restore standard bus speed after first attempt.
Fix #9153
closed 2023-07-12 by DzikuVx