iNavFlight/inav #9160: msc mode does not work again

lida2003, 2023-07-05:
## Current Behavior
It seems msc issue is back, see https://github.com/iNavFlight/inav/issues/5860

1. 6.1.1FW + inav configurator 6.1.0 + AOCODARCH7DUAL （Onboard Flash）  ==》 USB drive can't opend, no BB log, but it can save log as TXT file.
2. 

## Steps to Reproduce
1. Fly with blackbox enabled
2. blackbox tab, [save flash to file...] button works well, there is a txt log file
3. cli tab, [MSC] button works, but only usb drive "INAV_FC" appears, it can't be opened.

So we can't get BB log files.

## Expected behavior
Open USB drive, and copy BB log files to the computer.

## Additional context

Only below U driver appears, it can't be opened.

[image]

- FC Board name and vendor:  AOCODARCH7DUAL （Onboard Flash）
- INAV version string: 6.1.1 [AOCODARCH7DUAL]

carlybaby69, 2023-09-11:
Same issue on a Matek 765 Wing. Same SD card and same laptop with a Matek F405 Matek Wing msc command works fine.

DzikuVx, 2024-02-02:
Tested on INAV7 and H7 
[image]
 and everthing seems to be working. Slow and you have to wait circa 1m for files to appear, but works

PR #9572 (2023-12-16): USB MSC: Fix SCIS mode sense write protection bit
The USB MSC / SCSI layer currently doesn't set the write protect bit in the SCSI ModeSense response. Due to this, the Emulated FAT file system used for the internal flash doesn't get mounted read-only. When mounting the drive on Linux, the host attempts to write to the drive (I assume for the access time) , which fails and then the mounting doesn't succeed.

This fix sets the relevant bit in the SCSI ModeSense responses. The host doesn't attempt to write to the drive and the mounting works as expected.

This is only for H7 targets. The fix also [... cut]
closed 2024-02-02 by DzikuVx