iNavFlight/inav #9161: Presets for larger multirotors

RCnator, 2023-07-05:
## Current Behavior

Currently there is only presets for quads up to 7 inch.
## Desired Behavior

Since I guess the name Inav stands autonomous flight maybe there should be presets for larger quads and not only racing size quads. In my case 15 inch.
## Suggested Solution

Maybe add presets from 9 inch with 3 inch increments up to 18?

## Who does this impact? Who is this for?

Quad and multirotor users, all skill levels.
## Additional context

intelligentpotato, 2023-07-06:
+1 on this.
I believe iNav 5 had a preset for 10" which was a fine starting point for everything big and slow, but they removed it in iNav 6. Additionally, it seems to me that they have changed the PID controller logic/sensitivity in iNav 6. I flashed iNav 6 yesterday, migrated the iNav 5 settings that flew fine and crashed upon take off by flipping over. The thrust graph for that "flight" looks like god damn saw teeth. My PIDs are 90 34 54 - basically the old defaults.

MrD-RC, 2023-07-06:
You’re more than welcome to suggest actual settings that work with a range of different quad at those safe settings.

RCnator, 2023-07-06:
Unfortunately I myself lack the skill and know how to do that but hopefully someone else picks up the ball.

shota3527, 2023-08-08:
> You’re more than welcome to suggest actual settings that work with a range of different quad at those safe settings.

Agreed, we need a safe settings for wide range of quads.

agoliveira, 2023-08-21:
I would like to leave my +1 here. I just added a 10" to my fleet and I'm having a very hard time trying to make it work. It oscillates a lot. The funny thing is that I used inav 2.51 IIRC in a very large quad and worked quite well.

Dronitoni, 2023-09-11:
I currently try to find the best PID's for my 16-inch Quad. Seems not that easy in INAV6. 
7-incher works perfectly.

RCnator, 2023-11-12:
I can see Inav 7 is out now but there is no mentioning of improvements regarding this issue unfortunately.

MrD-RC, 2023-11-12:
> I can see Inav 7 is out now but there is no mentioning of improvements regarding this issue unfortunately.

No one has put forward any suggested setups. AFAIK, no developers have quads over 7”. As already stated, someone with larger quads will need to make suggested settings which are tested by the community and work on a wide range of quads with the target prop size.

TBH, there just doesn’t seem to be that much interest. Maybe start a conversation on the INAV discord’s multi-rotor channel?

MrD-RC, 2023-11-12:
I created the discussion on Discord. https://discord.com/channels/791437907478577172/1173188186382549032

petrweigl, 2024-03-02:
is there some progress on this issue?

MrD-RC, 2024-03-02:
There had been no preset recommendations. Also not a lot of interest on Discord. It seems like most people don’t fly quads that size.

petrweigl, 2024-03-02:
I have quadcopter with 12inch props running on inav7, trying to tune it. I am getting poor performance … with my settings.

RCnator, 2024-03-03:
A lot of us dont have a discord account, see above, and dont want to open up yet another online account, especially when we already have github. 

For Inav to live up to its name I think there should be a bit more focus on larger multirotors rather than racing quads, but I also remember when we were contested requesting dshot, [https://github.com/iNavFlight/inav/issues/1489]

MrD-RC, 2024-03-03:
The problem is there is simply not enough interest. And the INAV pilots who actually have these larger quads aren’t coming up with any suggestions for settings.  These settings will work for a large range of quads in that size category. 

So without any suggestions for settings, which are then tested by the community. The profiles cannot be added. If you guys with 10” quads want a preset. I’m sorry, but you’ll need to put in the legwork, experiment with the tunes, share findings with others to help come up with the settings. Without that. The presets you want won’t happen. The ball is in your court guys. 

I’m going to convert this to a discussion. As that is now more fittin.