iNavFlight/inav #9173: Use of hard tabs in code style ?

breadoven, 2023-07-12:
I've noticed there seems to be a lot of hard tabs used in PR's that have been merged recently. This screws up the formatting of the code when viewed on Github making it hard to read. I thought hard tabs were not to be used according to:

https://github.com/multiwii/baseflight/wiki/CodingStyle

Has this requirement been relaxed ?

MrD-RC, 2023-07-12:
Not as far as I know. Tabs should be used for indentation, not spaces.

breadoven, 2023-07-13:
You can use Tabs, just not hard tabs, soft Tabs should be used, i.e. padded with spaces. Your code editor should be set up to make a Tab 4 spaces, a soft tab. If you don't do this then the file viewer will tabulate the code according to its tabulation settings which may not correspond to 4 spaces. Using a soft Tab ensures consistency because a space should be consistent across viewers.

The affect of hard tabs can be seen here with the indentation a mess:
[image]
The arrows are hard Tabs:
[image]

It should look like this to avoid this problem, i.e. just spaces:
[image]

I seem to remember the Workflow checker used to fail if hard tabs were used. Obviously this doesn't happen any more.

MrD-RC, 2023-07-15:
Damn, just realised I'm a perp here! VSCode uses hard tabs by default. I've just changed the setting. In the settings, search for `Insert spaces`. It will return *Editor: **Insert Spaces*** Turn that off, and it stops with the hard spaces.

stronnag, 2023-07-15:
Unfortunately its just too easy to  go with `$EDITOR`'s default C-style. Mea culpa, hard tab fan and several other bracket related sins contrary to the style guide. I used to have a `.emacs` rule to put me on the straight and narrow, but ... just another offender.

mmosca, 2023-08-06:
> Unfortunately its just too easy to go with `$EDITOR`'s default C-style. Mea culpa, hard tab fan and several other bracket related sins contrary to the style guide. I used to have a `.emacs` rule to put me on the straight and narrow, but ... just another offender.

I am not against having .emacs .vimrc files in the repo to try to enforce desired behavior as default.

MrD-RC, 2023-08-22:
A nice little tip for VSCode

To convert existing indentation from spaces to tabs hit Ctrl+Shift+P and type: `Convert indentation to Tabs`

RomanLut, 2023-08-28:
Note: Please do not waste time fixing tabs in HITL-related code; they are fixed in PR: https://github.com/iNavFlight/inav/pull/8347
(sorry noticed tab problem too late)