iNavFlight/inav #9175: Incorrect throttle mid range with reversible motors

pguillory, 2023-07-14:
## Current Behavior
With reversible motors enabled, the throttle range to allow arming on my plane is 1550 to 1650.

[image]

## Steps to Reproduce
1. Apply the above settings for reversible motors.
2. Open the Receiver tab.
3. Slowly raise the throttle on the transmitter from minimum to maximum (988 to 2012 in my case).
4. Observe the FM telemetry sensor (from ELRS) change from `!ERR` to `OK` at 1550 and back to `!ERR` at 1650.

## Expected behavior
Based on the above settings, I would expect the throttle range for arming to be either 1500 +/- X (based on Reversible Motors Neutral), or maybe 1425 to 1575 (based on Reversible Motors Deadband Low/High). Seems like the range needs to include 1500, because that's the midpoint on my transmitter and the FC and the BLHeli_32 ESC.

## Suggested solution(s)
If it's expected behavior and not a bug, maybe a documentation or UI change is in order to make it more clear exactly where the throttle arming range comes from with reversible motors enabled.

## Additional context
I ran `dump` in the CLI and grepped for 1550 and 1600 and 1650, just trying to find the setting these thresholds are coming from, and that didn't lead anywhere.

https://pastebin.com/2NSmjmVm

- FC Board name and vendor:
```
SpeedyBee F405 WING
```
- INAV version string:
```
# version
# INAV/SPEEDYBEEF405WING 6.0.0 Apr 12 2023 / 18:10:46 (2e8837cc)
# GCC-10.2.1 20201103 (release)
```

pguillory, 2023-07-14:
Looking at calculateThrottleStatus() in rc_controls.c, I see the mid-throttle range is calculated as `PWM_RANGE_MIDDLE +/- mid_throttle_deadband`, where PWM_RANGE_MIDDLE is 1500 (I'm pretty sure?), and mid_throttle_deadband is the 3d_deadband_throttle config setting.

I tried changing 3d_deadband_throttle to 75 and tested the mid throttle arming range again, and found that it's now 1525 to 1675. So the effective formula is definitely 1600 +/- 3d_deadband_throttle. But... why 1600? Why not 1500?

pguillory, 2023-07-14:
Further progress. It looks like the throttle value from the receiver is run through a calculation involving the min_command, max_throttle, and throttle_idle settings before it is checked against the arming range. Since I recently changed throttle_idle to 0, I tried restoring it to the default of 15, and my apparent throttle arming range is now 1442 to 1618.

Also changed max_throttle from 1900 to 2000, now my throttle arming range is 1393 to 1549.

So at least I can arm now at 1500 throttle, which is nice. But now I realize that the throttle value from my transmitter is *not* the same as the throttle value going to the motor/ESC. For example if my transmitter says throttle 1600, the motor output on the Outputs tab says 1620. Or another example, transmitter throttle 1300 translates to motor throttle 1338.

I'm not sure how someone would have debugged this without reading the C++ source code. The issue is that with reversible motors, arming is based on an internal value `rcCommand[THROTTLE]`. It's potentially very different from the throttle value shown in the transmitter or the receiver tab, due to the calculation involving min_command/max_throttle/throttle_idle. It's also potentially very different from the motor output shown in the Outputs tab, due to 1) maybe differential thrust, 2) maybe power limiting, or 3) definitely until you arm the motor output is just 3d_neutral. So there's this internal value you can't see, that needs to fall into the arming range.

Well, the other issue is that the range is 1500 +/- 3d_deadband_throttle, which is not clear at all. It's not 3d_deadband_low/3d_deadband_high, and it's nothing involving 3d_neutral.

So, to recap:
1. Arming is based on this internal throttle command, rather than the throttle input or motor output. It makes sense that the distinct value exists, to account for power scaling and power limiting. However, if reversible motors is enabled and 3d_neutral is 1500, and the transmitter sends 1500, then the formula needs to work out such that `rcCommand[THROTTLE]` is also 1500. That not being the case seems like a bug to me. Power scaling should not be happening inside the deadband, right?
2. The value of the throttle command is not shown anywhere. This wouldn't be an issue, though, if the above is fixed.
3. The formula for the throttle command is not really documented and super complex. Also wouldn't be an issue if the midpoint issue were fixed.
4. The UI is very misleading about the range into which the throttle command needs to fall in order to enable arming.  It's *not* what you would think based on looking at the UI. Can the midpoint arming range just be 3d_deadband_low/3d_deadband_high, and we deprecate 3d_deadband_throttle? Something like that.

MrD-RC, 2023-07-14:
IMHO, throttle is messed up in INAV. Everything should be based on a 1000-2000 throttle range. It should only be scaled on output.

breadoven, 2023-07-14:
The issue here is that reversible motors use `THROTTLE_STATUS_TYPE_COMMAND `as the throttle value to determine "Throttle Low" which uses the mixed throttle value `rcCommand[THROTTLE]` rather than the raw RC stick position value. I cleaned up this code a while back:

https://github.com/iNavFlight/inav/pull/8601/files#diff-f092d782c82e8f36c4b664aacef2e02cc56c56640a837cc2a6205bc2397b0eed

Reversible motors have always used `THROTTLE_STATUS_TYPE_COMMAND `... not sure why because based on what you have found it doesn't make immediate sense. At the moment `THROTTLE_STATUS_TYPE_COMMAND `is only used in `calculateThrottleStatus()` to check for "Throttle Low". Using THROTTLE_STATUS_TYPE_RC instead would appear to work better and be more logical, i.e. reversible motor throttle low is simply 1500 +- the dead band. But you'd maybe need to double check what's happening in `mixer.c `to fully understand the reversible motor stuff because appears a bit complicated.

LsHallo, 2023-08-28:
setting `min_check` to 1000 seems to fix the weirdness.

TheBiggestBoon, 2024-03-26:
I have been having issues with this as well. My transmitter goes from 1000 to 2000 with 1500 being the midpoint. In iNav I have bidirectional enabled with 1500 being the midpoint and a deadband of 1475 to 1525. The motor spins backwards at 1500 and stops at 1535, then starts spinning forward at 1600. How do I get this centered around 1500 with the proper deadband? Id like to lower the deadband to +-5.

ama9910, 2024-05-19:
Ughh.. I discovered this bug today too. With deadband set to 1450-1500-1550 and all other settings default other than reversible motors and 0% idle throttle, I can't arm until about 1620. Setting min_check to 1000 seemed to make no difference.

Silentflyer, 2024-12-27:
Unfortunately, this issue has still not been resolved.
I helped myself by adjusting the PPM center of the gas channel in the remote control to 1600 instead of the usual 1500 for the middle position.
This is how it is possible to ARM well.
I also connected the brushless controller directly to the receiver and thus bypassed the flight control at this point, because when the throttle is controlled via the flight controller it is messy and jerky. 
I would be really happy if the ROVER topic in INAV received a little more attention from developers, because the trend for FPV ground vehicles is constantly increasing.
cheers

[image]

[image]

sercona, 2025-02-03:
I'm also having an issue similar to (or exactly like) this one.

I'm using inav 8 on a speedybee f405wing FC.  elrs rx and hoping to use on a rover with a single pwm bi-dir motor and a single steering servo.  really standard rc car stuff.

the only way I can get stability is to make the motor a servo and auto-arm inav (ie, set aux5 manually from 1000-2000 so that its always in arm mode).  this allows the motor output to be treated as throttle motor and it wont zoom to 1000 if you disarm.  disarming seems to force throttle to 1000.  so I worked around it by NOT using throttle (at all) and using 2 servo outputs, with one going to the motor esc and one going to the steer servo.

[image]

[image]

[image]

[image]

sercona, 2025-02-03:
diff is attached (its mostly generic, but its a config for 'rovers' on a SB f405wing)

[INAV_8.0.0_cli_20250202_184029.txt](https://github.com/user-attachments/files/18635615/INAV_8.0.0_cli_20250202_184029.txt)

sensei-hacker, 2025-02-03:
I have tested all default settings, other than enabling reversible motors. It arms just fine, with the throttle centered, when using default settings.  The output puts out Reversible Motors Neutral, as intended. This defaults to 1475.

If you are experiencing a problem while using other settings, **please attach the diff as requested**. We will not be able to find and fix the problem that occurs with your settings if we don't know what your settings are.

Please do not add another "me too" comment without a diff attached. Thank you.

sercona, 2025-02-03:
I said that the diff is my WORK AROUND and so there are no motors because motor behavior is not usable when you properly add motors.  I'll take a short demo video; that will explain much better what I'm seeing WHEN I try to use 1 servo and 1 motor.

sensei-hacker, 2025-02-03:
>  motor behavior is not usable when you properly add motors. 

As I mentioned, I works fine for me, using default settings.

I asked you four times for the diff that does not work.

> I'll take a short demo video; that will explain much better what I'm seeing WHEN I try to use 1 servo and 1 motor.

A video is cool and all; I can't do anything for you as long as you refuse to provide the diff. My settings work, yours don't.  To help you with your settings, I need to have your settings.

cyrilius42, 2025-05-24:
I've been following this thread and following the procedures everyone is using. In the end i could not get it working with my traxxas stampede 4x4 vxl using stock ESC connected to speedybee f405 wing mini. My short term solution was a Radiomaster ER6 output to steering and motor with "force telemetry off" in the webui and a secondary ELRS receiver (speedybee nano) to the FC for telemetry out to my Radiomaster pocket  WIFI backpack to my ground station for telemetry. I would like to figure out the configuration for the Radiomaster ER6 to output "Serial on UART 2" feature eventually to eliminate the secondary ELRS receiver. I'm very new to this hobby but would like to share my diff files to possibly get more insight if anyone is interested.

breadoven, 2025-05-24:
As mentioned above reversible motors use `rcCommand[THROTTLE]` rather than the raw throttle output from the transmitter which is used for normal motors. `rcCommand[THROTTLE]` output is affected by the `min_check `setting which has a default value of 1100 which changes the throttle output value significantly vs the raw value. It essentially reduces the useable throttle range from 1000 to 900 with the default value. So the 1500 midpoint becomes 444 (400/900) rather than 1500 causing the problems here.

It's perfectly OK to reduce `min_check `to a value closer to 1000 which should avoid the problems here because it will effectively make `rcCommand[THROTTLE]` range 1000 to 2000 rather than 1100 to 2000 using the default value of `min_check`. But you need to be aware that `min_check `(and `max_check`) is used to identify the stick end points for various functions including stick commands as well as low throttle so the setting used for `min_check `does need to be slightly higher than the minimum throttle output of the transmitter. For something like ELRS which has a minimum throttle value of 988 you could set `min_check `to 1000. For transmitters that output 1000 at min throttle you might want to use a `min_check `value of perhaps 1010. The only reason it's set to 1100 as I understand it is because of older systems that output channel values > 1000 with 100% endpoints set.

Looking at this stuff again there do seem to be various thinks that don't look right so it might need further work to sort it out properly.

cyrilius42, 2025-05-24:
ill try reducing it today and see what range for min will yield a better result. I noticed the 988 minimum throttle value for ELRS but never really knew the benefit/ detriment of having/ using it at that low of a end limit. I appreciate the quick reply and extra set of eyes to brainstorm with.

sensei-hacker, 2025-05-24:
>  I'm very new to this hobby but would like to share my diff files to possibly get more insight if anyone is interested.

Yeah that's what I've been asking for, for a few years now.

If someone would share a diff showing what is not working for them, I can help adjust things so it works.

sensei-hacker, 2025-05-26:
@cyrilius42  Can you please try flashing this file and see how it works for you?

https://drive.google.com/file/d/18DuLL0OpAKsolym82QxYedeeJFqGQiK0/view?usp=sharing

@breadoven  made a change which may help, but it needs to be tested.

sensei-hacker, 2025-05-27:
Thanks for the awesome test report!
I'll see if I can find anything further.

cyrilius42, 2025-05-27:
i noticed the 3d dead band throttle is in a scale of 200 with the setting at 50 (I'm assuming this is for a midpoint?) while the throttle mid is in a scale of 100 with a setting of 50. I have no idea what I'm looking at but i thought I'd mention it if it could be helpful.

breadoven, 2025-05-27:
A log would be useful then we can see exactly what the ESC is receiving from the FC. I assume this isn't one of those cases where the ESC needs to be calibrated via the Configurator outputs tab motor output rather than via the transmitter ?

cyrilius42, 2025-05-27:
> A log would be useful then we can see exactly what the ESC is receiving from the FC. I assume this isn't one of those cases where the ESC needs to be calibrated via the Configurator outputs tab motor output rather than via the transmitter ?

That was one of the issues i was having as well. In inav the "i understand" for motors in output will not allow me to control the esc from software. Almost as if its not receiving the proper communication. I've tried changing from standard to dshot 600 with varying results but none able to go through calibration. For log files I'm assuming you're referencing blackbox log files?

cyrilius42, 2025-05-28:
https://drive.google.com/drive/folders/1bxw2VSJtdvKW1ljpysD0k6DTUqgWuAfn?usp=sharing

I've attached a link with blackbox log files and a video with the log files starting at 1:31. Log 46 is the first with full range attempt at arming, second is after when i was rambling. I have a few more attempts with video where i botched the log process and accidentally wrote 45 different log files in one session but that footage has my attempt on inav trying the calibration from inav ( "i understand the risks" is not an option to slide for some reason). I can provide those logs and video if it helps, thanks again for your help and quick replies.

breadoven, 2025-05-28:
Well in the mid stick position it's sending 1475 to the motor rather than the 1500 you set with `3d_neutral` (I assume this is still correct ?). `rcCommand[THROTTLE]` is 1500 mid stick so that's not the problem, it's behaving as expected. I assume the ESC is expecting 1500 so won't calibrate ? Not sure where the 1475 is coming from, doesn't tie in with the code unless I'm missing something.

cyrilius42, 2025-05-30:
> > > Well in the mid stick position it's sending 1475 to the motor rather than the 1500 you set with `3d_neutral` (I assume this is still correct ?). `rcCommand[THROTTLE]` is 1500 mid stick so that's not the problem, it's behaving as expected. I assume the ESC is expecting 1500 so won't calibrate ? Not sure where the 1475 is coming from, doesn't tie in with the code unless I'm missing something.
> > 
> > 
> > That's on me, I neglected to set it back to 1500 before from a previous test. I'll record blackbox data again today at 1500 when I'm near the unit.
> 
> No need for more logging if `3d_neutral` was set to 1475 for that last logged test. If it was then things seem to work as expected in which case I'm not sure why it won't calibrate because the ESC should be receiving 1500 mid, 2000 high and 1000 low. Maybe it doesn't like the signal behaviour around the mid stick point where it's not linear. All I can suggest is calibrate it without the FC, directly from the Rx, and see how it works when the FC is included. Otherwise try playing with `3d_deadband_high `and `3d_deadband_low` see if that helps, i.e. try the values used at the top of this thread.

If i use a standard speedybee or bayck 2.4 elrs module directly to the flight controller i get this throttle/ calibration issue. When i use the radiomaster ER6 and connect the esc to output1 steering servo to output3, all is well so i keep looking towards the FC for an issue. I have two F405Wing mini FC and one F405Wing to test other firmware or hex files if you ever need help testing anything in Rover software. My next attempt will be ardurover to see if the issue happens on a different software, in an attempt to pinpoint the issue.

billyd60, 2025-06-29:
Reversible throttle hasn't worked correctly since v4.1 The problem is it requires you to arm in full reverse, or you only get to use half the range of throttle if you want to arm at throttle neutral (1500).

closed 2025-08-14 by breadoven