iNavFlight/inav #9182: Remaining Flight Distance values missing in OSD

b14ckyy, 2023-07-20:
## Current Behavior
OSD shows no values for the remaining flight distance. No dashes or any character at all. 

## Steps to Reproduce
1. Set up necessary values for remaining flight time and distance
2. Enable both items in OSD
3. Fly the plane
4. Remaining flight time values are shown but for distance there is no value, just the icon. 

## Expected behavior
Remaining Flight distance shown in OSD

## Suggested solution(s)
none

## Additional context
- Not checked with digital systems yet. 
- Fonts loaded but no change (Impact in my case)
- Pixel and Character OSD affected
- need to check previous versions of INAV

some screenshots: 
[image]
[image]

---

- FC Board name and vendor: Matek F722-WPX, Matek F411-WSE (Possibly all)
- INAV version string: INAV 6.1.1 Release

JulioCesarMatias, 2023-07-20:
A check logic was broken after this commit:
https://github.com/iNavFlight/inav/commit/e0840a4ae6e7784ec46ccff8c61e47b1576dce90#diff-da6a9ff6562212212bb06f79de4b84480c7c52e772d3708498a21a640d8fd7aaR213

See this OSD line:
https://github.com/iNavFlight/inav/blob/366db3a347a3c0c965d505519093f2f848178396/src/main/io/osd.c#L2125

The verification logic of "ARMING_FLAG(ARMED)" in True and False state are checked at the same time in:

`if ((!ARMING_FLAG(ARMED)) || (distanceMeters == -1)) {
            buff[4] = SYM_BLANK;
            buff[5] = '\0';
            strcpy(buff + 1, "---");`

MrD-RC, 2023-07-21:
@breadoven Would this line not need to be `!ARMING_FLAG(ARMED)`.  Surely if should be calculating the wind if armed, not disarmed?

https://github.com/iNavFlight/inav/blob/e0840a4ae6e7784ec46ccff8c61e47b1576dce90/src/main/flight/rth_estimator.c#L213

breadoven, 2023-07-21:
> @breadoven Would this line not need to be `!ARMING_FLAG(ARMED)`. Surely it should be calculating if armed, not disarmed?
> 
> https://github.com/iNavFlight/inav/blob/e0840a4ae6e7784ec46ccff8c61e47b1576dce90/src/main/flight/rth_estimator.c#L213

The ! is for both` (FIXED_WING_LEGACY || ARMED)`, i.e. if not a plane and not armed then return. So if you're armed and a plane it shouldn't return so should work as required when armed (but not when disarmed). Having said that it's not right. If it was a multirotor for instance it wouldn't return as it should when armed. It should have been a `&&` rather than a `||`. Negated `||`'s seem a bit confusing for some reason.

JulioCesarMatias, 2023-07-21:
I think the style of development should change in relation to this... The use of '!((condition) or (condition))' could no longer be accepted, and should it always be revised by the merger to only accept the type: '((!condition) or (!condition))'

breadoven, 2023-08-24:
Is anyone working on a fix for this or is something else being worked on to get around the general issue of `char `being limited to a byte causing problems for higher value symbols ?

MrD-RC, 2023-08-24:
@breadoven I have a working fix. Just need to submit the PR.

PR #9333 (2023-10-04): Flight remaining flight distance value
Tested by @b14ckyy 

Fixes #9182
closed 2023-10-04 by MrD-RC