iNavFlight/inav #9186: Asking for support for the TMOTORVELOXF7V2 board.

masterguy29, 2023-07-21:
## Current Behavior
The current iNav software does not support TMOTORS Velox Cine boards.

## Desired Behavior
I would like for there to be a firmware that supports the TMOTORS Velox Cine boards.

## Suggested Solution
Add a target for the TMOTORVELOXF7V2. There seems to be alot of similarities between the TMOTORF7V2 and TMOTORVELOXF7V2 boards so it should be pretty easy to add the new target made specifically for the Velox boards to help lower incompatibilities.

## Who does this impact? Who is this for?
All users.

## Additional context

MrD-RC, 2023-07-21:
You need to ask T-Motor. Here is a link to the new hardware policy https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md

masterguy29, 2023-07-21:
> You need to ask T-Motor. Here is a link to the new hardware policy https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md

I have contacted tmotors and asked them to collaborate with iNav devs to help with this project. Is that all I needed to do? Or do I have to send in a test board as well?

mmosca, 2023-07-25:
> 

Ideally, tmotor should send hardware, but if you really want support, you could provide hardware yourself.

Item 2: https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md#new-hardware-support

```Requester is a community member not affiliated with a manufacturer of the hardware. Requester provides hardware samples to at least one core developer and the firmware support already exists in official Cleanflight or Betaflight firmware.```

masterguy29, 2023-07-27:
> Ideally, tmotor should send hardware, but if you really want support, you could provide hardware yourself.
>
> Item 2: [...]

T-Motor has reached out to me and advised me that they will go ahead and collaborate with iNav to get support for the Velox F7. Love to see it!

closed 2023-07-27 by masterguy29