iNavFlight/inav #9188: Rangefinder Sensor (VL53L1X) not working an Speedybee F7V3

maxledfreak, 2023-07-22:
## Current Behavior
 Sensor VL53L1X on i2C is not detected. Measurement with logic analyzer shows no request to sensor. To validate this behaviour the following procedure were done:

1. Measurment with no rangefinder sensor selected
[image]

Measurement:
[image]

2. Measurment with rangefinder sensor selected (VL53L1X)
[image]
Measurement:
[image]

## Steps to Reproduce

1.Reboot FC

## Expected behavior
 Sensor is deteced with address 0x76 (7 bit address)

## Suggested solution(s)
 Maybe mapping to all ragefinder sensors is connected to internal bus like Acc., Baro etc.

## Additional context

---

# version
# INAV/SPEEDYBEEF7V3 6.0.0 Mar 22 2023 / 11:23:04 (084101ac)
# GCC-10.2.1 20201103 (release)

- FC Board name and vendor:
- INAV version string: SPEEDYBEEF7V3

druckgott, 2023-07-27:
https://github.com/iNavFlight/inav/pull/9200

maxledfreak, 2023-08-01:
@DzikuVx @druckgott fix is not working

druckgott, 2023-08-06:
Can be closed, fixed an tested with @maxledfreak 
https://github.com/iNavFlight/inav/pull/9221

closed 2023-08-10 by mmosca

mxchinist, 2023-09-18:
I've had issues making the VL53L1X (bought on aliexpress) work with INAV (version 5.1.0.)
To make it work, I had to add this into the target.h file (inav-release_x.x.x\src\main\target\YOURTARGETNAME\target.h) before compiling :

#define USE_RANGEFINDER
#define RANGEFINDER_I2C_BUS     BUS_I2C1
#define USE_RANGEFINDER_VL53L1X

(your FC might be using BUS_I2C2 if it has one, check where is the magnetometer configured, that's where most likely will the rangefinder work)

I also had to check the sensors address (all the aliexpress VL53L0X and VL53L1X sensors seems to have 0x29 address).
And set up the correct address in inav-release_x.x.x\src\main\target\common_hardware.c (in this case the 0x29) :

#if defined(USE_RANGEFINDER_VL53L1X)
    #if !defined(VL53L1X_I2C_BUS) && defined(RANGEFINDER_I2C_BUS)
        #define VL53L1X_I2C_BUS RANGEFINDER_I2C_BUS
    #endif

    #if defined(VL53L1X_I2C_BUS)
    BUSDEV_REGISTER_I2C(busdev_vl53l1x,     DEVHW_VL53L1X,      VL53L1X_I2C_BUS,    0x29,               NONE,           DEVFLAGS_USE_RAW_REGISTERS,  0);
    #endif

Then the sensor, after connecting to 5V,GND,SCL and SDA pins (SDA->SDA, SCL->SCL)* on your FC should work.