iNavFlight/inav #9193: Add Target For HOBBYWING XRotor F7 Convertible Flight Controller.

ThePrevious1, 2023-07-25:
Hi guys! Is there any chance of adding a target for the Hobbywing  XRotor F7 FC? The standalone version of this flight controller comes with a barometer and IC2 pads so I was pretty sure myself that I was gonna find it in the target list the day I decided to flash iNav. To my surprise there wasn't a target for it.

Here's the Betaflight dump in case it comes in handy. Thanks all of you for the hard work you guys do to provide us with this amazing free software :) 

board_name HOBBYWING_XROTORF7CONV
manufacturer_id HOWI

# name: -

# resources
resource BEEPER 1 C14
resource MOTOR 1 B01
resource MOTOR 2 B04
resource MOTOR 3 B03
resource MOTOR 4 A15
resource MOTOR 5 C08
resource MOTOR 6 C09
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource I2C_SCL 1 B06
resource I2C_SCL 2 NONE
resource I2C_SCL 3 NONE
resource I2C_SCL 4 NONE
resource I2C_SDA 1 B07
resource I2C_SDA 2 NONE
resource I2C_SDA 3 NONE
resource I2C_SDA 4 NONE
resource LED 1 C15
resource LED 2 NONE
resource LED 3 NONE
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_SCK 4 NONE
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MISO 4 NONE
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
resource SPI_MOSI 4 NONE
resource ADC_BATT 1 C01
resource ADC_RSSI 1 C02
resource ADC_CURR 1 C00
[... 924 lines cut ...]
set feedforward_averaging = OFF
set feedforward_smooth_factor = 25
set feedforward_jitter_factor = 7
set feedforward_boost = 15
set feedforward_max_rate_limit = 90
set dyn_idle_min_rpm = 0
set dyn_idle_p_gain = 50
set dyn_idle_i_gain = 50
set dyn_idle_d_gain = 50
set dyn_idle_max_increase = 150
set level_race_mode = OFF
set simplified_pids_mode = RPY
set simplified_master_multiplier = 100
set simplified_i_gain = 100
set simplified_d_gain = 100
set simplified_pi_gain = 100
set simplified_dmax_gain = 100
set simplified_feedforward_gain = 100
set simplified_pitch_d_gain = 100
set simplified_pitch_pi_gain = 100
set simplified_dterm_filter = ON
set simplified_dterm_filter_multiplier = 100
set tpa_mode = D
set tpa_rate = 65
set tpa_breakpoint = 1350

closed 2023-10-27 by DzikuVx

muhammedoney, 2024-12-24:
@DzikuVx hello,
I need the same target also.
Is the problem is solved in issue #7343 ?

sensei-hacker, 2024-12-24:
The best course of action is to contact the manufacturer.

https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md