iNavFlight/inav #9195: Altitude and speed scroll bars move the wrong way.

malinhiles, 2023-07-25:
## Current Behavior
https://github.com/iNavFlight/inav/assets/93371446/3a9c15cc-85eb-4dbc-b2bd-6e218bc5e2e5

The altitude and speed scrolls currently move inverse of how they should. Here's a quick dvr clip where my airspeed reduces and my altitude increases.

## Expected behavior
https://youtu.be/5cPoxfRWeV4?t=288
Here is a clip of a Garmin autopilot's flight computer with the actual altitude and speed scrolls that iNav is trying to emulate. This clip shows the correct directions for how the bars should be scrolling.

malinhiles, 2023-08-26:
Any thoughts on this?

b14ckyy, 2024-03-22:
For years I found it logical that altitude up > bar goes up.... 
But now that you say it, I can't unthink this.... imho you are totally right and the altitude bar direction makes no sense xD

In the FrSky Pixel OSD its correct: https://youtu.be/V1L5fFvsOGU?t=493

sensei-hacker, 2026-09-27:
Hello from the future!
Three years later, this has been changed / fixed in https://github.com/iNavFlight/inav/pull/12025

closed 2026-09-27 by sensei-hacker