iNavFlight/inav #9197: accelerometer and gyro not working SPEEDYBEEF405V3

ebawywkawy, 2023-07-26:
I got a new FC (speedybeef405v3) and my gyro and accelerometer are not working with all versions of INAV (stable and release-candidate).
Appreciate your help!

- FC Board name and vendor: SPEEDYBEEF405V3 
- INAV versions: 6.1.1, 6.1.0, 6.1.0-RC1, 6.0.0, 6.0.0-RC3, 6.0.0-RC2, 6.0.0-RC1, 6.0.0-FP1, 5.1.0.

[image]

mmosca, 2023-07-28:
Does it work with bf? Sounds like there may have been a imu change, or you have a bad board.

ebawywkawy, 2023-07-28:
> Does it work with bf? Sounds like there may have been a imu change, or you have a bad board.

Yes, i have no problem with betaflight. Speedybee F405 V3 has Bmi270. Сhanging the accelerometer in INAV didn't help.

closed 2024-02-02 by DzikuVx