iNavFlight/inav #9198: Add TMOTOR VELOX CINE F7 target

dengdizzy, 2023-07-27:
Please add TMOTOR VELOX CINE F7 target.
I can provide the MCU SCH and samples to help you make target.

Here is the link
https://tmotorhobby.com/goods-1248-TMOTOR+Velox+Cine+F7+Multifunctional+Freestyle+Cinematic++Flight+Controller+For+FPV+Drones.html

mmosca, 2023-07-28:
@dengdizzy are you affiliated with TMotor?

dengdizzy, 2023-07-31:
> @dengdizzy are you affiliated with TMotor?

Sure.

mmosca, 2023-07-31:
> > @dengdizzy are you affiliated with TMotor?
> 
> Sure.

In that case, please have a look at: 

https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md

There is an e-mail in there you can use to contact the team to sort out the details.

dengdizzy, 2023-08-02:
> > > @dengdizzy are you affiliated with TMotor?
> > 
> > 
> > Sure.
> 
> In that case, please have a look at:
> 
> https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md
> 
> There is an e-mail in there you can use to contact the team to sort out the details.

> > > @dengdizzy are you affiliated with TMotor?
> > 
> > 
> > Sure.
> 
> In that case, please have a look at:
> 
> https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md
> 
> There is an e-mail in there you can use to contact the team to sort out the details.

Thank you,i will close this issue.

closed 2023-08-02 by dengdizzy