iNavFlight/inav #9202: using CrazyF405HD ELRS with INAV

sgrizzi, 2023-07-29:
I am trying to program a CrazyF405HD ELRS to run with INAV. The FC is working fine with Betaflight, so I guess I can assume that the hardware blocks are working.
Happymodel state that this board is compatible with INAV and that I should use BetaflightF4 as a target, and I successfully loaded this firmware after a full erase of the chip.
The on board ELRS receiver is working fine.
Unfortunately I cannot get the gyro (BMI270 with SPI) to work - it looks as if the gyro is expected to use i2c in INAV, but I may be wrong.
Can anybody give some hints on how to get the gyro to work, or point me to a tutorial/documentation on how to program this board with INAV?
Thanks a lot!

sgrizzi, 2023-07-29:
Looking at the sw code, it would appear that a SPEEDYBEEF405V3 would be a better target board to use. Is this right?

ebawywkawy, 2023-07-30:
Hello sgrizzi,

i have the same issue #9197 with my SPEEDYBEEF405V3. Gyro (BMI270) does not work.

sgrizzi, 2023-07-30:
Hi ebawywkawy,
well, I just tried to program my CrazyF405 board using SPEEDYBEEF405V3 as a target, latest version (6.1.1) and something does work.
BMI270 gyro/accelerometer, BMP280 barometer, the onboard ELRS receiver do work well.
I did not do anything special. The BMI270 is recognized at startup and I could reset and calibrate the gyro/acc normally.

On the other hand I cannot program the ESCs: the esc-configurator fails to read the ESCs parameters (and this does work using Betaflight). Do you happen to know if there is any passthrough issue with the ECSs???

closed 2023-08-01 by sgrizzi

dezelvol, 2023-12-09:
@sgrizzi have you succeeded in configuring the board? 
I have the same problem. By the way, on Inav v7 it's not connecting to the configurator at all. 
On BetaflightF4 v6.1 no sensors are accessible, on SPEEDYBEEF405V3 v6.1 there are sensors but no ESC connection.

sgrizzi, 2023-12-10:
Yes, I did manage to load INAV 6.1 on the CrazyF405 using a modified BetaflightF4 target board.
The target board configuration "as-is" is not working properly and I had to recompile the code modifying the setup as follows:
-fixed the use of the BMI270 gyro/acc to be used instead of the default MPU6000
-fixed the BMP280 barometer connection which is using the SPI bus instead of the I2C
In my setup I also added a WS-M181 GPS, connecting the magnetometer to the i2c bus in parallel to the connections to the BMP280.

Hope this can help.
Rgds.

rallasballas, 2024-06-11:
Could you share your compiled code? I don't think I have a clue to make the changes myself. I would like to try INAV on my GoFilm20 which uses this same all in one board.
Also, where did you connect the magnetometer I2C to? I couldn't find any I2C pads.

iolkhovskyi, 2024-06-12:
I did it long ago, so I'm unsure if it's the correct file. Besides, I don't advise installing firmware obtained from strangers. 

[inav_6.1.1_BETAFLIGHTF4.zip](https://github.com/user-attachments/files/15800749/inav_6.1.1_BETAFLIGHTF4.zip)

dezelvol, 2024-06-14:
I did it long ago, so I'm unsure if it's the correct file. Besides, I don't advise installing firmware obtained from strangers.

[inav_6.1.1_BETAFLIGHTF4.zip](https://github.com/user-attachments/files/15838616/inav_6.1.1_BETAFLIGHTF4.zip)

mxchinist, 2024-10-20:
> I did it long ago, so I'm unsure if it's the correct file. Besides, I don't advise installing firmware obtained from strangers.
> 
> [inav_6.1.1_BETAFLIGHTF4.zip](https://github.com/user-attachments/files/15838616/inav_6.1.1_BETAFLIGHTF4.zip)

This soft-bricks the board, definitely not the correct firmware. I advise naming the files accordingly, to prevent such things.

dezelvol, 2024-10-20:
I hope you were able to unbrick board. I'm sure it's possible as I've tested all versions myself. 

I'll remove file to avoid anyone using it.

breadoven, 2025-12-16:
Ah ... missed that. Didn't check the unstable releases in Configurator and haven't updated from master for a while. Shame the PR didn't reference this Issue. 

https://github.com/iNavFlight/inav/pull/10989