iNavFlight/inav #9204: NEUTRONRCF435MINI.AIO

Guzgog, 2023-07-31:
I have installed inav_6.1.1_NEUTRONRCF435MINI.hex on to the AIO version of this board, on a Quad, all seems to be working fine with GPS Mag etc etc, except Motor2 will not spin, other three motors work as expected. I looked at Resources and Motor 2 is allocated to A10 Whereas in the Betaflight target M2 is set to B07. The other three motors are allocated the same as the BF target.

## Desired Behaviour
Need motor 2 operate, 

## Suggested Solution
I presume motor 2 allocated to B07.

## Who does this impact? Who is this for?
If this could be addressed, then I will test fly and if all is well a new target for NEUTRONRCF435MINI.AIO can be added?

## Additional context
In BF and Inav the other motors are: Motor1-B01, Motor3-A03, Motor4-B01.

DusKing1, 2023-07-31:
No, you should not check any other project if you are going to use INAV on NEUTRONRCF435MINI. The target file has no problem. Motor 2 is A10 instead of B07.

shanggl, 2023-08-07:
this is a bug : changed the pin defines before batch production , this pr fix it #9226  @Guzgog @DusKing1

closed 2023-08-07 by DzikuVx

wahyudius, 2023-10-13:
have @anyone have to fix it or costum hex for this FC