iNavFlight/inav #9217: acoda-Rc f4 v2 no accelerometer dected

naseraliomran, 2023-08-05:
**PLEASE MAKE SURE YOU READ AND UNDERSTAND THE SOCIAL MEDIA SUPPORT CHANNELS. QUESTIONS ABOUT FLASHING, CONFIGURING, PILOTING MAY BE CLOSED WITHOUT FURTHER INTERACTION.**

* [INAV Discord Server](https://discord.gg/peg2hhbYwN)
* [INAV Official on Facebook](https://www.facebook.com/groups/INAVOfficial)
* [INAV Official on Telegram](https://t.me/INAVFlight)

**Please double-check that nobody reported the issue before by using search in this bug tracker.**

**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****

## Current Behavior

the acc and the gyr is not detected  ihave MPU 6500 it says in the cli unavailable 
## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

snaewe, 2023-09-12:
I have the same problem. @naseraliomran Does your FC look like this:

[image]

i.e. does your version have the MPU6000 IMU ?

snaewe, 2023-09-13:
@sensei-hacker 

[image]

sensei-hacker, 2023-09-13:
Thanks. I was curious because you had referenced the V1 issue and wrote your board model as "FC Board name and vendor: Acoda f4 v1.0", but changed the V2 target. I see your board is marked V2.

sensei-hacker, 2023-09-13:
@naseraliomran  Sorry I didn't see this issue before. Can you please confirm whether your board silkscreen says V1, V1.1, or V2?

I believe I tested the target on a V1.1.

closed 2024-02-02 by DzikuVx