iNavFlight/inav #9218: GPS Autobaud fails on M8 (and possibly older GPS)

stronnag, 2023-08-06:
Several users of `master` builds have reported that their GPS is not recognised in current `master` (at least `fd59159f` 2023-08-03). 

Disabling `auto_baud` appears to fix this, so possibly #9174 is not safe on older GPS that can only cope with 230400 hardwarily (speculation).

With auto-baud == OFF:
```
# status
INAV/MATEKF405 7.0.0 Aug  3 2023 / 07:00:03 (fd59159f)
GCC-10.3.1 20210824 (release)
System Uptime: 388 seconds
Current Time: 2023-08-06T07:51:11.812+00:00
Voltage: 0.00V (0S battery - NOT PRESENT)
CPU Clock=168MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280, MAG=HMC5883
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
SD card: Startup failed
Stack size: 6144, Stack address: 0x10010000, Heap available: 1828
I2C Errors: 0, config size: 9028, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = none
  AIRSPEED : configured = none, used = none
System load: 8, cycle time: 1008, PID rate: 992, RX rate: 9, System rate: 9
Arming disabled flags: NAV RX CLI
OSD: MAX7456 [30 x 16]
VTX: not detected
GPS: HW Version: UBLOX8 Proto: 18.00 Baud: 115200
  GNSS Capabilities:
    GNSS Provider active/default
    GPS 1/1
    Galileo 1/0
    BeiDou 0/0
    Glonass 1/1
    Max concurrent: 3
```
```
# get gps
ahrs_gps_yaw_windcomp = ON
Allowed values: OFF, ON

gps_provider = UBLOX7
Allowed values: NMEA, UBLOX, UBLOX7, MSP, FAKE

gps_sbas_mode = EGNOS
Allowed values: AUTO, EGNOS, WAAS, MSAS, GAGAN, NONE

gps_dyn_model = AIR_1G
Allowed values: PEDESTRIAN, AIR_1G, AIR_4G

gps_auto_config = ON
Allowed values: OFF, ON

gps_auto_baud = OFF
Allowed values: OFF, ON

gps_ublox_use_galileo = ON
Allowed values: OFF, ON

gps_ublox_use_beidou = OFF
Allowed values: OFF, ON

gps_ublox_use_glonass = ON
Allowed values: OFF, ON

gps_min_sats = 6
Allowed range: 5 - 10

gps_ublox_nav_hz = 10
Allowed range: 5 - 200

inav_use_gps_velned = ON
Allowed values: OFF, ON
```

With `auto_baud` enabled, `status` reports GPS as `NONE`.

stronnag, 2023-08-06:
@mmosca  : Confirmed that reverting #9174 also fixes auto-baud on M8

krizw, 2023-08-06:
With Inav 7 and auto baud set to ON, the GPS is "unavailable", and set to OFF, back to normality. Same issue on both as below.
Matek F405 with Flywoo GM10 PRO V3
Flywoo F745 with HGLRC GPS M80PRO.

INAV/MATEKF405 7.0.0 Jul 27 2023 / 18:04:38 (33e2865e)
GCC-10.3.1 20210824 (release)
System Uptime: 896 seconds
Current Time: 2023-08-06T09:02:05.636+00:00
Voltage: 0.00V (0S battery - NOT PRESENT)
CPU Clock=168MHz, GYRO=MPU6500, ACC=MPU6500, MAG=QMC5883
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=NONE, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
SD card: Startup failed
Stack size: 6144, Stack address: 0x10010000, Heap available: 1828
I2C Errors: 0, config size: 9028, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = none, used = none
System load: 8, cycle time: 519, PID rate: 1926, RX rate: 9, System rate: 9
Arming disabled flags: ANGLE RX CLI
OSD: MAX7456 [30 x 16]
VTX: not detected
GPS: HW Version: UBLOX10 Proto: 34.09 Baud: 115200
  GNSS Capabilities:
    GNSS Provider active/default
    GPS 1/1
    Galileo 1/1
    BeiDou 0/0
    Glonass 1/0
    Max concurrent: 3

mmosca, 2023-08-06:
> @mmosca : Confirmed that reverting #9174 also fixes auto-baud on M8

I can have a look if the ultra high bauds can be made to work with m8, or if we can try to only touch the ultra high baud rates if the gps is set to something higher than 230400.

mmosca, 2023-08-06:
#9219 should get m8 working again.

I will probably try to make some extra changes there.

@stronnag can you test it with your old m8 gps as well?

stronnag, 2023-08-06:
That was "fun". Allowing auto-baud on the old m8 (where its port was previously set to 115200) resulted in the device resetting to 38400, which took a little while to resolve. Further experiments with mwptool's `ublox-cli` (which pretty much apes INAV's GPS handling) confirms that  this particular piece of junk hardware resets to 38400 if set to autobaud 
above 115200.

If it works, it spews out some diagnostics and then PVT...
```
$ ublox-cli -b 38400
2023-08-06T13:01:39+0100 Registered USB device: /dev/ttyUSB0 [10c4:ea60], Vendor: Silicon_Labs, Model: CP2102_USB_to_UART_Bridge_Controller, Serial: 0001, Driver: cp210x
/dev/ttyUSB0@38400
Data size 92b (1 7)
PVT: lat: 0.000000 lon: 0.000000 elev (msl/elpsd): -17.00 0.00 acc(h/v): 4294967.3/3750026.0
sats: 0, fix 0
GPS: 2013-09-01 00:12:20.000 - Host: 2023-08-06 12:01:39.689
Auobaud 0
Auobaud 1
Auobaud 2
Auobaud 3
Auobaud 4
Rate initialised 38400
Data size 92b (1 7)
PVT: lat: 0.000000 lon: 0.000000 elev (msl/elpsd): -17.00 0.00 acc(h/v): 4294967.3/3750026.2
sats: 0, fix 0
GPS: 2013-09-01 00:12:20.600 - Host: 2023-08-06 12:01:40.289
Disable NMEA
Request version
Data size 100b (a 4)
Version info after 0.035972s (len 100b)
SW: [2.01 (75331)] HW: 00080000 80000
ExtVer:  PROTVER 15.00
ExtVer:  GPS;SBAS;GLO;BDS;QZSS
Data size 92b (1 7)
PVT: lat: 0.000000 lon: 0.000000 elev (msl/elpsd): -17.00 0.00 acc(h/v): 4294967.3/3750026.2
sats: 0, fix 0
GPS: 2013-09-01 00:12:20.800 - Host: 2023-08-06 12:01:40.489
Set air 1G
send INAV ublox v7 init [80000] PVT
Data size 92b (1 7)
PVT: lat: 0.000000 lon: 0.000000 elev (msl/elpsd): -17.00 0.00 acc(h/v): 4294967.3/3750026.2
sats: 0, fix 0
GPS: 2013-09-01 00:12:21.000 - Host: 2023-08-06 12:01:40.717
Data size 104b (1 30)
SVINFO: Channels 8
```
```
$ ublox-cli -b 115200
2023-08-06T13:02:02+0100 Registered USB device: /dev/ttyUSB0 [10c4:ea60], Vendor: Silicon_Labs, Model: CP2102_USB_to_UART_Bridge_Controller, Serial: 0001, Driver: cp210x
/dev/ttyUSB0@115200
Auobaud 0
Auobaud 1
Auobaud 2
Auobaud 3
Auobaud 4
Rate initialised 115200
Disable NMEA
Data size 92b (1 7)
PVT: lat: 0.000000 lon: 0.000000 elev (msl/elpsd): -17.00 0.00 acc(h/v): 4294967.3/3750027.8
sats: 0, fix 0
GPS: 2013-09-01 00:12:43.800 - Host: 2023-08-06 12:02:03.471
Request version
Data size 100b (a 4)
Version info after 0.012411s (len 100b)
SW: [2.01 (75331)] HW: 00080000 80000
ExtVer:  PROTVER 15.00
ExtVer:  GPS;SBAS;GLO;BDS;QZSS
Set air 1G
Data size 92b (1 7)
PVT: lat: 0.000000 lon: 0.000000 elev (msl/elpsd): -17.00 0.00 acc(h/v): 4294967.3/3750027.8
sats: 0, fix 0
GPS: 2013-09-01 00:12:44.000 - Host: 2023-08-06 12:02:03.671
send INAV ublox v7 init [80000] PVT
Data size 92b (1 7)
PVT: lat: 0.000000 lon: 0.000000 elev (msl/elpsd): -17.00 0.00 acc(h/v): 4294967.3/3750027.8
sats: 0, fix 0
GPS: 2013-09-01 00:12:44.200 - Host: 2023-08-06 12:02:03.883
Data size 128b (1 30)
SVINFO: Channels 10
Chn	ID	used	healthy	quality
  1	G3	-	Y	1
  2	G4	-	Y	3
  7	G9	-	Y	1
  5	G10	-	Y	1
  3	G12	-	Y	3
  0	G13	-	Y	1
  4	G17	-	Y	1
  9	G26	-	Y	1
  6	----	-	-	1
 10	----	-	-	3
Set rate 5 Hz
Set SBAS
Data size 92b (1 7)
PVT: lat: 0.000000 lon: 0.000000 elev (msl/elpsd): -17.00 0.00 acc(h/v): 4294967.3/3750027.8
sats: 0, fix 0
GPS: 2013-09-01 00:12:44.400 - Host: 2023-08-06 12:02:04.070
Data size 92b (1 7)
PVT: lat: 0.000000 lon: 0.000000 elev (msl/elpsd): -17.00 0.00 acc(h/v): 4294967.3/3750027.8
```
```
$ ublox-cli -b 230400
2023-08-06T13:02:25+0100 Registered USB device: /dev/ttyUSB0 [10c4:ea60], Vendor: Silicon_Labs, Model: CP2102_USB_to_UART_Bridge_Controller, Serial: 0001, Driver: cp210x
/dev/ttyUSB0@230400
Auobaud 0
Data size 92b (1 7)
PVT: lat: 0.000000 lon: 0.000000 elev (msl/elpsd): -17.00 0.00 acc(h/v): 4294967.3/3750029.3
sats: 0, fix 0
GPS: 2013-09-01 00:13:05.800 - Host: 2023-08-06 12:02:25.471
Auobaud 1
Auobaud 2
Auobaud 3
Auobaud 4
Rate initialised 230400
Disable NMEA
Request version
Set air 1G
send INAV ublox v6 init [0] POSLLH
Set Neo 6 TimeUTC
Set rate 5 Hz
Set SBAS
### now we're screwed, back to 38400 ....
```

mmosca, 2023-08-06:
Well, nice to run into old hardware quirks :)

Good news is I have some changes locally that would let you set max auto baud to 115200.

I haven't tested it with hardware yet, but pushed the changes anyway.

stronnag, 2023-08-06:
OK again here. Might be a crappy ublox clone, but it's sitting in doors with 7 satellites.

mmosca, 2023-08-06:
> OK again here. Might be a crappy ublox clone, but it's sitting in doors with 7 satellites.

Is that with the latest changes and setting gps_auto_baud_max to 115200?

stronnag, 2023-08-06:
> Is that with the latest changes and setting gps_auto_baud_max to 115200?

It is. 
And for completeness, we need a `PG_REGISTER_WITH_RESET_TEMPLATE(gpsConfig_t, gpsConfig, PG_GPS_CONFIG, **)` version bump. Everyone forgets that.

mmosca, 2023-08-06:
> > Is that with the latest changes and setting gps_auto_baud_max to 115200?
> 
> It is. And for completeness, we need a `PG_REGISTER_WITH_RESET_TEMPLATE(gpsConfig_t, gpsConfig, PG_GPS_CONFIG, **)` version bump. Everyone forgets that.

I always forget it :)
 I will add it.

PR #9219 (2023-08-06): Restore autobaud for m8 gps
Fixes  #9218 

Add gps_auto_baud_max with a default setting of 230400, matching INAV 6.0 behavior.

Newer GPS units may be set to higher baud rates and some old units seem to have bugs causing them to reset and revert 38400 if you attempt to auto baud higher than 115200.
closed 2023-08-06 by mmosca