iNavFlight/inav #9233: FOXEER F722 V4 X8 firmware not working

simoccc, 2023-08-11:
## Current Behavior

Brand new Foxeer F722 V4 X8 with MPU6000 Gyro/acc and DPS310 Baro.

( https://www.foxeer.com/foxeer-f722-v4-mpu6000-fc-8s-dual-bec-barometer-x8-g-502 )

Inav 6.1 Firmware not recognizing Baro ( wrong type ) and Gyro ( unavailable ) and wrong UART numbering :

# status
INAV/FOXEERF722V4_X8 6.1.1 Jun 13 2023 / 10:25:34 (42ced253)
GCC-10.2.1 20201103 (release)
System Uptime: 16 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 0.00V (1S battery - NOT PRESENT)
CPU Clock=216MHz, BARO=SPL06
STM32 system clocks:
  SYSCLK = 216 MHz
  HCLK   = 216 MHz
  PCLK1  = 54 MHz
  PCLK2  = 108 MHz
Sensor status: GYRO=UNAVAILABLE, ACC=UNAVAILABLE, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE
Stack size: 6144, Stack address: 0x20010000, Heap available: 1840
I2C Errors: 10, config size: 9009, max available config: 16384
ADC channel usage:
   BATTERY : configured = ADC 1, used = none
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = none
  AIRSPEED : configured = none, used = none
System load: 12, cycle time: 504, PID rate: 1984, RX rate: 9, System rate: 9
Arming disabled flags: ANGLE CAL HWFAIL RX CLI DSHOTBEEPER
OSD: not enabled
VTX: not detected

## Steps to Reproduce

Flash firmware - status cli - ports tab only with 5 part instead of 6 

## Suggested solution(s)

## Additional context

Tried to flash Foxeer F722 V4 firmware , UART are correct but Gyro/acc not recognized...

# status
INAV/FOXEERF722V4 6.1.1 Jun 13 2023 / 10:24:17 (42ced253)
GCC-10.2.1 20201103 (release)
System Uptime: 93 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 0.00V (1S battery - NOT PRESENT)
CPU Clock=216MHz, BARO=SPL06
STM32 system clocks:
  SYSCLK = 216 MHz
  HCLK   = 216 MHz
  PCLK1  = 54 MHz
  PCLK2  = 108 MHz
Sensor status: GYRO=UNAVAILABLE, ACC=NONE, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE
Stack size: 6144, Stack address: 0x20010000, Heap available: 1840
I2C Errors: 5, config size: 9033, max available config: 16384
ADC channel usage:
   BATTERY : configured = ADC 1, used = none
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = none
  AIRSPEED : configured = none, used = none
System load: 12, cycle time: 516, PID rate: 1937, RX rate: 9, System rate: 9
Arming disabled flags: ANGLE CAL HWFAIL RX CLI
OSD: not enabled
VTX: not detected

IgnacioHR, 2023-08-12:
Hi, I managed to sort this out by downloading source from tag 6.1.1, modifying provided driver and local compiling. iNav version 7.0.0 already has the fix

simoccc, 2023-08-12:
Thanks for the feedback , so no fix for 6.1 I guess I need to wait 7

IgnacioHR, 2023-08-12:
Here is my build for version 6.1.1 use at your own risk!

[inav_6.1.1_FOXEERF722V4.hex.zip](https://github.com/iNavFlight/inav/files/12327069/inav_6.1.1_FOXEERF722V4.hex.zip)

closed 2023-10-27 by DzikuVx