iNavFlight/inav #9237: Improve how to read QMC5883 compass

IgnacioHR, 2023-08-13:
## Current Behavior

In case of incorrect readings of the compass after it was initialised the values are reset to 0. this might cause strange reactions to the drone.

## Steps to Reproduce

I discovered this problem just reading the source code and trying to figure out what could be the reason for spikes in the yaw while using heading hold. It is not demonstrated that my hardware is not right, but failing to read the compass will, for sure, cause a strange movement if we don't fix it.

## Expected behavior
The drone shall assume no changes from the previous good reading of the compass. IMHO this is safer. Also the driver should annotate the number of errors internally and this could be obtained later for inspection.

## Suggested solution(s)

Change the code more or less like this. I've not created a pull request until someone else could tell me if I'm right or not.

[image]

## Additional context

As stated before, I'm having some spikes in yaw while heading hold is activated and I'm digging into the source code to evaluate possible reasons. It might be my problem is unrelated with reading the compass but my code proposal is just to mitigate the possibility an incorrect reading of the compass generates strange drone reacctions.

---

INAV/FOXEERF722V4 6.1.1 Aug  5 2023 / 16:33:43 (42ced253)
GCC-10.2.1 20201103 (release)

IgnacioHR, 2023-08-13:
Hmm...

In case of a failure read, the values are 0s. on file imu.c, in function `static void imuCalculateEstimatedAttitude(float dT)` around line 762 there is this code
```
    fpVector3_t measuredMagBF = {.v = {mag.magADC[X], mag.magADC[Y], mag.magADC[Z]}};
```

That is considering the 0s as good values!

shota3527, 2023-10-31:
Hi 
This PR will reject invalid compass data, such as magnetic field is too strong or too weak
https://github.com/iNavFlight/inav/pull/9387

closed 2024-03-22 by b14ckyy