iNavFlight/inav #9240: TcmmRC f4 with gyroscope bmi270

Flightnew, 2023-08-15:
**PLEASE MAKE SURE YOU READ AND UNDERSTAND THE SOCIAL MEDIA SUPPORT CHANNELS. QUESTIONS ABOUT FLASHING, CONFIGURING, PILOTING MAY BE CLOSED WITHOUT FURTHER INTERACTION.**

* [INAV Discord Server](https://discord.gg/peg2hhbYwN)
* [INAV Official on Facebook](https://www.facebook.com/groups/INAVOfficial)
* [INAV Official on Telegram](https://t.me/INAVFlight)

**Please double-check that nobody reported the issue before by using search in this bug tracker.**

**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****

## Current Behavior

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

Flightnew, 2023-08-15:
Hi, I have a Aliexpress TcmmRc F4  with gyroscope bmi270 or bmi160 (can't define exactly).
On BETAFLIGHT the board is called Matek f405, but INAV do not have such a board. The gyroscope is not detected on the INAV Matek f405 firmware. Is there any board I could choose to flash INAV on it to make everything work.

Tanks

MrD-RC, 2023-08-15:
Duplicate https://github.com/iNavFlight/inav/issues/9241#issuecomment-1678373776

closed 2023-08-15 by MrD-RC