iNavFlight/inav #9241: TcmmRC f4 com giroscópio bmi270

Flightnew, 2023-08-15:
## Current Behavior

## Desired Behavior

## Suggested Solution

## Who does this impact? Who is this for?

## Additional context

Flightnew, 2023-08-15:
Hi, I have a Aliexpress TcmmRc F4 with gyroscope bmi270 or bmi160 (can't define exactly).
On BETAFLIGHT the board is called Matek f405, but INAV do not have such a board. The gyroscope is not detected on the INAV Matek f405 firmware. Is there any board I could choose to flash INAV on it to make everything work.

MrD-RC, 2023-08-15:
Sounds like a clone board who are piggybacking off the original target. The problem is when the changes components to those not supported by the original hardware.

What should really happen is that TcmmRc should submit their own target for their flight controller, and maintain that. https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md

Flightnew, 2023-08-15:
olá, ela utiliza o firmwre da Matek 405. Existe a TcmmRc F4 com mpu 6000

closed 2023-10-27 by DzikuVx