iNavFlight/inav #9247: Sending commands to drone in the X-Plane simulator with inav SITL

NikitaYeryomin, 2023-08-17:
### Discussed in https://github.com/iNavFlight/inav/discussions/9246

<div type='discussions-op-text'>

<sup>Originally posted by **NikitaYeryomin** August 17, 2023</sup>
Hello INAV community,

I'm currently in the process of developing an autopilot system, with the decision-making software running on a Jetson Nano. My primary objective is to have the drone react in a simulator based on commands sent from my script.
I'm using the following setup:

1. X-Plane 11 simulator from Steam on Windows 10 with NK_FPVSurfwing model from the [INAV-X-Plane-HITL](https://github.com/RomanLut/INAV-X-Plane-HITL) repo. I have configured Data output to the 49000 port. Settings can be viewed below: [image]
I've also enabled all [datarefs](https://res.cloudinary.com/disqgapnj/image/upload/v1692259009/xplane_data_output2_l7vfbs.jpg) I found in the [xplane.c](https://github.com/iNavFlight/inav/blob/master/src/main/target/SITL/sim/xplane.c) code of the inav SITL.

2. inav_SITL running on the same host in WSL (Ubuntu 20.04) with the following command line:
`inav_SITL --sim=xp --simip=192.168.0.162 --simport=49000 --chanmap=S01-01,S02-02,S03-03,M01-04`
It's able to successfully connect to X-Plane: [image]

3. INAV Configurator 6.1.0 - successfully connects the inav_SITL on port 5760 and I can see it receives the data updates from the simulator: [image]
Here are also my ports and receiver configuration tabs: 
[image]
[image]
 
To communicate with the inav_SITL with my script I've created [this fork](https://github.com/NikitaYeryomin/pyMultiWii) of pyMultiWii library. It uses TCP socket instead of serial to connect to inav_SITL and send MSP commands. I'm able to successfully connect to inav_SITL on port 5761 and retrieve any data like attitude or RC data, e.g.:

``````from pymultiwii import MultiWii
import time

def get_rc_data(board):
    rc_data = board.getData(MultiWii.RC)
    return rc_data

if __name__ == "__main__":
    host = '127.0.0.1'
    port = 5761
    board = MultiWii(host, port)
    
    while True:
        try:
            rc_data = get_rc_data(board)
            if rc_data is not None:
                print(rc_data)
            else:
                print("Could not retrieve RC data.")
        except Exception as e:
           print(f"Error: {e}")
        time.sleep(0.1)
``````

    However, I'm not able to send any commands to the drone, like arming or changing throttle. Here's the code I've tried to use for take off:

``````
def take_off(drone, takeoff_throttle, duration):
    if not drone.arm():
        print("Failed to arm the drone!")
        return

    print("Taking off...")
    start_time = time.time()

    hover_throttle = 1500  
    elevon_neutral = 1500 

    start_time = time.time()
    while time.time() - start_time < duration:
        # Set throttle for takeoff and maintain elevon in neutral position
        data = [elevon_neutral, elevon_neutral, 1500, takeoff_throttle]
        drone.sendCMD(8, MultiWii.SET_RAW_RC, data, '4H')
        time.sleep(0.1)

    # Set throttle to hover after takeoff
    data = [elevon_neutral, elevon_neutral, 1500, hover_throttle]
    drone.sendCMD(8, MultiWii.SET_RAW_RC, data, '4H')
    print("Hovering...")
``````

    It does not respond with any error but looks like it just ignores them as I don't see any visual changes in the simulator as well as it does not affect any RC data values. 

Has anyone faced a similar issue or can provide guidance on what might be going wrong?

Any help or insight much appreciated!</div>

RomanLut, 2023-08-20:
I think you have to set receiver type to "MSP" to use SET_RAW_RC.

Also make sure you set "RSSI Channel" to f.e. 15 in recevier tab and pass 2000 in this channel, otherwise inav will not arm because he thinks that RSSI is 0.

stronnag, 2023-08-21:
It is not necessary to assert RSSI.

It is necessary to:

* Ensure the SITL EEPROM / FC is configured correctly (MSP RX (or emulation of any other supported serial RX) , sensors, switches, modes)
* Ensure the SITL FC is in an arming state.

The following [fl2sitl](https://github.com/stronnag/bbl2kml/wiki/fl2sitl) session shows the SITL going through these steps (with MSP RX).

## fl2sitl

Note:

* Checking configuration, including `map`
* Getting BOX Ids and correlating switches, ranges and flight modes
* Checks arming status as FC does calibrations, this takes time ...
* Around 5 seconds for calibrations to complete to "Ready to Arm"

 ```
[fl2sitm] 10:32:15.356462 ******** Opened RX **************
[fl2sitm] 10:32:15.456995 Serial init
[SOCKET] [::1]:51188 connected to UART1
[fl2sitm] 10:32:15.557539 Serial running 6
[fl2sitm] 10:32:16.796217 INAV v7.0.0 SITL (bef8acca) API 2.5
[fl2sitm] 10:32:16.799953 nav_extra_arming_safety: 0 (bypass false)
[fl2sitm] 10:32:16.800510 Status from rx 0
[fl2sitm] 10:32:16.800525 RC MAP AETR
[fl2sitm] 10:32:16.801028 "BENCHYMCTESTY"
[fl2sitm] 10:32:16.802171 ARM;PREARM;ANGLE;HORIZON;TURN ASSIST;HEADING HOLD;CAMSTAB;NAV POSHOLD;LOITER CHANGE;NAV RTH;NAV WP;HOME RESET;GCS NAV;WP PLANNER;MISSION CHANGE;NAV CRUISE;NAV COURSE HOLD;SOARING;NAV ALTHOLD;MANUAL;NAV LAUNCH;SERVO AUTOTRIM;AUTO TUNE;AUTO LEVEL TRIM;BEEPER;BEEPER MUTE;OSD OFF;BLACKBOX;KILLSWITCH;FAILSAFE;CAMERA CONTROL 1;CAMERA CONTROL 2;CAMERA CONTROL 3;OSD ALT 1;OSD ALT 2;OSD ALT 3;
[fl2sitm] 10:32:16.802586 Serial init completed, with modes
[fl2sitm] 10:32:16.802609     {boxid:11 chanidx:0 start:16 end:32} PosHold
[fl2sitm] 10:32:16.802756     {boxid:10 chanidx:0 start:32 end:48} RTH
[fl2sitm] 10:32:16.802792     {boxid:28 chanidx:1 start:16 end:48} WP
[fl2sitm] 10:32:16.802809     {boxid:2 chanidx:2 start:16 end:32} Horizon
[fl2sitm] 10:32:16.802831     {boxid:3 chanidx:2 start:32 end:48} AltHold
[fl2sitm] 10:32:16.802849     {boxid:45 chanidx:2 start:32 end:48} Cruise3D
[fl2sitm] 10:32:16.802868     {boxid:36 chanidx:3 start:19 end:28} Launch
[fl2sitm] 10:32:16.802892     {boxid:27 chanidx:3 start:36 end:48} Failsafe
[fl2sitm] 10:32:16.802915     {boxid:0 chanidx:5 start:24 end:48} Arm
[fl2sitm] 10:32:16.802933     {boxid:12 chanidx:6 start:22 end:48} Manual
[fl2sitm] 10:32:16.802950     {boxid:13 chanidx:7 start:28 end:48}
[fl2sitm] 10:32:16.803032 RC init done
[fl2sitm] 10:32:16.804149 Boxflags: 80000 Armflags: Calibrate Overload NavUnsafe RCLink (0x40e20)
[fl2sitm] 10:32:16.904125 Boxflags: 80000 Armflags: Calibrate NavUnsafe RCLink (0x40a20)
[fl2sitm] 10:32:17.509233 Boxflags: 80000 Armflags: Calibrate NavUnsafe (0xa20)
[fl2sitm] 10:32:18.875456 Boxflags: 80000 Armflags: Calibrate (0x220)
[fl2sitm] 10:32:20.796932 Boxflags: 0 Armflags: Ready to arm (0x20)
[fl2sitm] 10:32:20.796940 Set status 1 (20)
[fl2sitm] 10:32:20.796948 Status from rx 1
[fl2sitm] 10:32:20.796952 ** Ready to arm (Press 'A' to arm) **
```

## msptest monitor

If we monitor the status (Configurator, [msptest](https://github.com/stronnag/msp-rs) etc.), we see arming is OK and enabled:

Also shows Arming OK

```
$ msptest  tcp://localhost:5760

                                MSP Test Viewer
                 v0.32.0 on Arch Linux 6.4.11-zen2-1-zen x86_64

Port    : localhost:5760
MW Vers : ---
Name    : BENCHYMCTESTY
API Vers: 2.5 (MSP v2)
FC      : INAV
FC Vers : 7.0.0
Build   : Aug 17 2023 12:09:38 (bef8acca)
Board   : SITL
WP Info : 0 of 120, valid false
Uptime  : 12s
Power   : 16.7 volts,  0 amps
GPS     : fix 2, sats 16, 43.328484° -5.681429° 0m, 0m/s 0° hdop 1.00
Arming  : OK
IO Stats: 21660 messages in 10.9s (1990.2/s) (unknown: 0, crc 0)
Debug   : ---
```

There is no issue here, the SITL behaves correctly; [fl2sitl](https://github.com/stronnag/bbl2kml/wiki/fl2sitl)  demonstrates the same use case.

closed 2023-08-21 by stronnag