iNavFlight/inav #9253: Getting failsafe from Joystick as a RC

deeph46, 2023-08-20:
Hi,

I'm using OpenHd and combine with Inav. I made the OpenHD to used Joystick as control (RC) with Mavlink protocol. Everything works great, except the failsafe in Inav. I used Logitech G29 as a Joystick. When I simulate failsafe, I turning off the Ground Station power and then the throttle stay at last position not 1000. So it means it will run away. 

I'm using Inav as a Rover.

How do I get Failsafe at 1000 from Throttle channel?

Please let me know.

Thanks.

RomanLut, 2023-08-28:
OpenHD air unit is probally just continues sending last values. Pass RSSI in some channel. Otherwise INAV does not know that failsafe has occured.

ZhouYixun, 2023-08-31:
> So it also can't do some Programming (in Programming tab) to be able signal back to 1000?

Can Logitech x56 be used with openhd and inav? Looking forward to your reply

deeph46, 2023-09-01:
> > So it also can't do some Programming (in Programming tab) to be able signal back to 1000?
> 
> Can Logitech x56 be used with openhd and inav? Looking forward to your reply

As long as the RC only 1 USB connected to GS, it will detect. I connected G29 included the pedal, it detected 16ch I believe. But not all buttons are registered in OpenHD.

ZhouYixun, 2023-09-01:
So if it is a rocker plus a pedal, a total of two USB devices, gs can recognize? Won't you ignore one of them? Only some buttons don't work

ZhouYixun, 2023-09-01:
@deeph46

deeph46, 2023-09-01:
> So if it is a rocker plus a pedal, a total of two USB devices, gs can recognize? Won't you ignore one of them? Only some buttons don't work?

no, I have tried 2 USB devices plugged into GS, GS will use the first one plugged in. That was my experienced though. I still don't know why some buttons didn't work, but it's not a big deal for me, as I only use couple buttons. 1 for arming and 1 for other mode.

DzikuVx, 2023-10-27:
The is a FAILSAFE flight mode that triggers failsafe proceude. Configure it however you want

closed 2023-10-27 by DzikuVx