iNavFlight/inav #9263: Burned 4 motors on testing land failsafe procedure

axelisart, 2023-08-26:
## Current Behavior
I wanted to test Failsafe Land Procedure. So I took off at around 1 meter and activate Failsafe via RC switch. Copter fall down, flip over due to impact to the ground, then and only then the motors started to spin for landing procedure. As the copter was upside down in grass, propellers could not spin. So I tried to disarm copter to avoid motor burn. Unfortunately I was not able to disarm despite activating the right RC switch for that. Consequently all motors burned due to the landing procedure that activated throttle during 20 seconds and motors couldn't spin...

## Steps to Reproduce

1. Set FailSafe to Land Procedure
2. Set Throttle Value used while Landing to 1250us
3. Set Delay for turning off motors during failsafe to 200ds (default value)
4. Set one RC channel to FailSafe
5. Take off in stabilized mode
6. Go to 1 meter altitude
7. Activate FailSafe
8. Try to disarm

## Expected behavior
Copter should switch to Land procedure and Land. I was hopping that when I activated FailSafe mode, the motors instantaneously changed to Failsafe throttle value and that then that the copter landed. Instead motor stopped for a short time before trying to spin again. There was a latency like as long as one or two seconds before motors activate again to the throttle value parameter set in FailSafe screen (I set it to 1250us). 

## Suggested solution(s)
Copter should change instantaneously to throttle value set in FailSafe Land Procedure. There should not be any latency during the transition between throttle level at the failsafe time and throttle level set in FailSafe land parameter. In that case, even if copter is near the ground at the FailSafe time, the landing will occurs correctly.
It should also be possible to disarm during FailSafe Procedure. It would add some security to be able to stop propellers spinning in case of landing in wrong place (and would have save my 4 motors...)

- FC Board name and vendor: Kakute F7 HDV
- INAV version string: 6.1.1

## Additional context
Have no logs to provide due to no SD at the time of testing.

Jetrell, 2023-08-26:
Do you have `gyro_use_dyn_lpf = ON` ?

breadoven, 2023-08-27:
FailSafe set to Land Procedure will only use the set throttle value if estimated altitude isn't available. If estimated altitude is available (only requires a working Baro)  then an auto rate controlled descent will be used instead with the rate set by `nav_emerg_landing_speed` which is 5m/s by default. The reason the motors appeared to stop is probably because the quad was trying to achieve the required descent rate of 5 m/s which effectively causes it to drop initially. At only 1 m above the ground it wouldn't have had time to start controlling although even if it had 5m/s is quite a high rate of descent, probably better to use 2m/s. Not sure why this was set to a default of 5m/s, perhaps just a compromise to get down as quickly as possible with a reasonable chance of minimal damage on hitting the ground.

Disarming via switch is prevented during Failsafe to avoid inadvertent disarm caused by unpredictable Aux channel behaviour. A Kill switch will still work though.

axelisart, 2023-08-28:
Hi, thanks for replies. I better understand the behavior of my copter now. Maybe that this Land procedure should include a flip detection (rotation of 90deg or more in X/Y axis during land) that activate a kill or disarm in order to avoid copter motor/esc damages.

breadoven, 2023-08-28:
> Hi, thanks for replies. I better understand the behavior of my copter now. Maybe that this Land procedure should include a flip detection (rotation of 90deg or more in X/Y axis during land) that activate a kill or disarm in order to avoid copter motor/esc damages.

That's worth thinking about. I can't see any normal reason a quad would be inverted for any length of time with the motors running.

DouglasKooChenSoon, 2023-09-24:
> FailSafe procedure = LAND is only a fall back, based on the old way Cleanflight used to work. It uses a set throttle value and a timer to control landing. And isn't very reliable.. Its always best set the Failsafe procedure = RTH.
> 
> If you want to test how the multicopter lands. It is best to perform an RTH via the RTH mode switch, with LAND_ALWAYS selected.. Or another method, is to takeoff and switch in to Poshold mode. Then once its hover in a fixed position, toggle the Poshold mode switch ON and OFF 5 times. Then the mulitcopter will initiate and Emergency Landing.. Make sure disarm on landing is enabled if you want the craft to disarm after landing.
> 
> Do you have `gyro_use_dyn_lpf = ON` ? If so, turn it OFF in 6.1.1

THIS is what I need for my long range Quad !! ( Really appreciate )

May i know if we trigger the "Emergency landing" by " toggling the Poshold Mode switch ON and OFF for 5 times ",     let's say the Quad is still in the air hovering (altitude 100 meter) at 100 meter distance away from home ,   in case where we couldn't have enough battery power to safely fly the drone back to Home (due to unexpected gust) or Motor Bearing failure ( which i couldn't detect it during my preflight checks )  ,       will this "Emergency Landing" (toggling 5 times Poshold Mode)  switches to RTH mode while the drone is descending in emergency landing and the RC signal is blocked by Trees and Terrains ( I had set my RC failsave in INAV as RTH)  ?

Jetrell, 2023-09-24:
> will this "Emergency Landing" (toggling 5 times Poshold Mode) switches to RTH mode while the drone is descending in emergency landing and the RC signal is blocked by Trees and Terrains ( I had set my RC failsave in INAV as RTH) ?

This was the pull request for that feature, for you to read through.

https://github.com/iNavFlight/inav/pull/8686

closed 2023-10-27 by DzikuVx