iNavFlight/inav #9290: FOXEER v4  mpu6000.  dji

y957128156, 2023-09-11:
**PLEASE MAKE SURE YOU READ AND UNDERSTAND THE SOCIAL MEDIA SUPPORT CHANNELS. QUESTIONS ABOUT FLASHING, CONFIGURING, PILOTING MAY BE CLOSED WITHOUT FURTHER INTERACTION.**

* [INAV Discord Server](https://discord.gg/peg2hhbYwN)
* [INAV Official on Facebook](https://www.facebook.com/groups/INAVOfficial)
* [INAV Official on Telegram](https://t.me/INAVFlight)

**Please double-check that nobody reported the issue before by using search in this bug tracker.**

**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****

## Current Behavior

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

y957128156, 2023-09-11:
Open the DJIFPV VTX port to display the aircraft battery level. Opening the MSP Displayport does not display the aircraft battery level. The receiving device is the latest firmware of DJI goggles2

y957128156, 2023-09-11:
[image]

mmosca, 2023-09-11:
Can you post a diff all and a screenshot of your ports tab?

y957128156, 2023-09-12:
[inav6.1.1.txt](https://github.com/iNavFlight/inav/files/12581547/inav6.1.1.txt)
[BF4.42.txt](https://github.com/iNavFlight/inav/files/12581548/BF4.42.txt)

y957128156, 2023-09-12:
This is the file of the same flight controller, it works fine in bf

y957128156, 2023-09-12:
> 您可以发布所有差异和端口选项卡的屏幕截图吗？

[image]

closed 2023-09-29 by DzikuVx