iNavFlight/inav #9293: STM32H723 support

youprint, 2023-09-14:
Hi everyone
I made a custom flight controller based on the STM32H723VGH, it's a recent MCU pin compatible with the STM32H743VIH already supported but faster (550 Mhz)
I tested it on Ardupilot and it's working great but I using mainly INAV for my planes.
I already tried to use the H743 target but it's not working since it is not the same clock configuration, and few other changes.
Betaflight already have a target for this MCU too.

mmosca, 2023-09-15:
It is probably doable, but we would probably need hardware to test with.

https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md

But if you would like to add support to it, it is probably relatively easy, since it is similar to the h743 which is already supported.

My guess is that you need to update the STM32H7 libraries in:

lib/main/STM32H7

Adn the system_stm32h7xxx.* files in src/main/targets

And have a look at the cmake folder for possible changes needed.

It may even kind of work if yo specify the target as target_stm32h723xg after you update the libraries.

youprint, 2023-09-15:
Thanks for your reply
My prototype board is working with Betaflight and Ardupilot now, I made it work yesterday at 500mhz.
Unfortunately I have only 3 boards so I can't send it to all the inav dev but I can do it myself I think.
Betaflight is very similar to Inav so you are right I need to update the H7 library like they did in Betaflight.
let's try....

mmosca, 2023-09-15:
If you manage to get it work, we would gladly accept the pull request updating the libraries and adding the infrastructure for the new mcu

youprint, 2023-09-22:
I managed to get it work @400mhz
I had a lot of issues if I try higher frequency, I'm using CubeMX to help me with the clock configuration but 400mhz is enough for me for now.
Manufacturers are starting to use 723/730 family now (SPRACING H7RF for exemple)
I think we will need that support in the near future...
Thank you for your help

mmosca, 2023-09-27:
Fell free to submit a pull request with the basics to support the H723/H730.
We can try asking Dominic for help/tips once the PR is in place.

hydra, 2024-02-06:
> I managed to get it work @400Mhz I had a lot of issues if I try higher frequency, I'm using CubeMX to help me with the clock configuration but 400mhz is enough for me for now. Manufacturers are [...]

Use 520Mhz, anything higher requires setting option bits and also disables error correction on the internal RAM.  Refer to the clock configuration that I used for H730 support in Betaflight.

H723/H730/H733/H725/H735 are all excellent choices for FC MCUs.  Low internal flash memory variants (e.g H730) can be used with memory mapped mode when paired with 133Mhz flash and the OCTOSPI interface but this brings additional programming challenges.

NiranjanRkanada, 2024-12-05:
hey guys even i need some guide can any one help me with that , can i use  STM32H743 flight controller firmware to STM32H723

NiranjanRkanada, 2024-12-05:
Can you provide me firmware

JoseAntonioMG, 2025-04-14:
Hi @youprint, 

Could you please send me the clock configuration in INAV for the STM32H723VGT?

I mean:

pllConfig_t pll1ConfigRevV = {
.clockMhz = 400,
.m = xx,
.n = xx,
.p = xx,
.q = xx,
.r = xx,
.vos = PWR_REGULATOR_VOLTAGE_SCALE0
};

b14ckyy, 2026-09-22:
Will close since we have not heard from the OP for 3 years now.

closed 2026-09-22 by b14ckyy