iNavFlight/inav #9296: JHEMCU GF16-BMI F405 gyro problem

NaturalCopters, 2023-09-16:
**PLEASE MAKE SURE YOU READ AND UNDERSTAND THE SOCIAL MEDIA SUPPORT CHANNELS. QUESTIONS ABOUT FLASHING, CONFIGURING, PILOTING MAY BE CLOSED WITHOUT FURTHER INTERACTION.**

* [INAV Discord Server](https://discord.gg/peg2hhbYwN)
* [INAV Official on Facebook](https://www.facebook.com/groups/INAVOfficial)
* [INAV Official on Telegram](https://t.me/INAVFlight)

**Please double-check that nobody reported the issue before by using search in this bug tracker.**

**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****

## Current Behavior

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

NaturalCopters, 2023-09-16:
Hello, I purchased the JHEMCU GF16-BMI stack F405. I want to get it to work under INAV, but it's having trouble finding the right Gyro.     The 3D model in Inav also shows wrong movement / direction.                  
I tried the 5.0.0 and the 6.1.0. firmware.
 
https://github.com/jhemcu/FC-ESC-Firmware/tree/main/GF16-BMI%2B13A%20ESC

NaturalCopters, 2023-09-16:
@pawelchcki Hello, I purchased the JHEMCU GF16-BMI stack F405. I want to get it to work under INAV, but it's having trouble finding the right Gyro. The 3D model in Inav also shows wrong movement / direction.
I tried the 5.0.0 and the 6.1.0. firmware.

https://github.com/jhemcu/FC-ESC-Firmware/tree/main/GF16-BMI%2B13A%20ESC

NaturalCopters, 2023-09-17:
https://youtu.be/cfqICV5ZcB8

mmosca, 2023-09-28:
INAV does not have an official target for this fc and I don't see the target source files on that repository either. So we can't even help you, if we try.

You can try contacting JHEMCU.

And also, refer to the new hardware policy:
https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md

closed 2023-09-28 by mmosca