iNavFlight/inav #9301: Smooth take off and landing.

Lalbabu2001, 2023-09-19:
## Current Behavior

When you try arm the quad in poshold or althold mode turned on it doesn't arm.
## Desired Behavior

Put quad in althold or poshold mode then arm the quad raise throttle to mid nothing happen , now as raise throttle above mid quad starts to climb at a predetermined rate put throttle again in mid quad maintain that position, then low throttle below mid quad descends at a predetermined rate.
## Suggested Solution

You can modify the althold or poshold code or insert a new mode under mode tab

## Who does this impact? Who is this for?

This is just for multirotors right now and even can be implemented in fixed wings. This is very helpful for beginners who learns to fly quad.  I don't wanted every time fly couple meters in angle mode and then switch into altitude mode and let the quad move to hower throttle and again adjust height using sticks. Not economical
## Additional context

b14ckyy, 2023-09-28:
This is intentional. Arming in active navigation modes can be very dangerous. 
You can still arm in angle mode then then switch to POSHOLD and lift of as you want. 

alternatively use INAV programming. just force INAV into ANGLE mode when disarmed with a channel override, its just 2 lines in programming. When armed it will switch to any mode you have set.

Lalbabu2001, 2023-09-29:
It's not dangerous because if there were enough satellites and everything
ok only then inav will allow take off in poshold mode like althold mode
does.

b14ckyy, 2023-09-29:
its still dangerous. If any GPS mode is active, INAV constantly corrects its position. All the time. 
Si if you arm and the electrical noise of the quad causes any kind of position deviation the quad will try to correct that but can't because it sits on the ground. So it can flip over and get out of control. To avoid that, its in general not allowed to arm with enabled nav modes. 

As I said: if you want that, there is a workaround with a few clicks but its on your own responsibility.

Lalbabu2001, 2023-09-29:
Then how my dji panthom does this?

b14ckyy, 2023-09-29:
It does the same. When you power it on the position controller is not active. if you lift off, it suddenly jumps up to  certain altitude in self level with just altitude hold and THEN activates position control for exactly that reason. you can do the same with logic if you want. Just add a sonar or LIDAR and set it up yourself. 

Maybe a developer will show interest in this change or this feature that would only work with additional hardware (baro is NOT reliable enough for smooth takeoff and landing). In my opinion there is no need for extra code to implement a launch function just because someone does not want to save one single switch movement after arming.

closed 2024-03-22 by b14ckyy